chipsalliance/Surelog #3554: When UHDM is bundled - Syrelog.pc breaks

yurivict, 2023-03-29:
```
$ pkg-config --cflags Surelog
Package UHDM was not found in the pkg-config search path.
Perhaps you should add the directory containing `UHDM.pc'
to the PKG_CONFIG_PATH environment variable
Package 'UHDM', required by 'Surelog', not found
```

UHDM.pc should be installed for the bundled UHDM.

hzeller, 2023-03-30:
When UHDM is bundled (i.e. you compile it from third_party/), the install process will install both, Surelog.pc and UHDM.pc.  So if your pkg-config has found Surelog.pc, it should also find UHDM.pc

```
PREFIX=/tmp/testinstall make install
$ find /tmp/testinstall -name "*.pc"
/tmp/testinstall/lib/pkgconfig/Surelog.pc
/tmp/testinstall/lib/pkgconfig/UHDM.pc
```

With the above install, you can also test this:
```
PREFIX=/tmp/testinstall make test_install_pkgconfig
```

Important is that if you have installed things in a non-standard location, to set your `PKG_CONFIG_PATH` search path

yurivict, 2023-03-30:
Unfortunately, only one .pc file is installed: Surelog.pc

hzeller, 2023-03-30:
that should only happen if you have `-DSURELOG_USE_HOST_UHDM=On` (because then, the UHDM.pc is already on your machine).

Do you have a log of the full command line you give to cmake and its output ?

yurivict, 2023-03-30:
```
===>  Configuring for surelog-1.50
===>  Performing out-of-source build
/bin/mkdir -p /usr/ports/cad/surelog/work/.build
cd /usr/ports/cad/surelog/work/.build; /usr/bin/env PYTHON="/usr/local/bin/python3.9" XDG_DATA_HOME=/usr/ports/cad/surelog/work  XDG_CONFIG_HOME=/usr/ports/cad/surelog/work  XDG_CACHE_HOME=/usr/ports/cad/surelog/work/.cache  HOME=/usr/ports/cad/surelog/work PATH=/usr/local/libexec/ccache:/usr/ports/cad/surelog/work/.bin:/home/yuri/bin:/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/sbin:/usr/local/bin PKG_CONFIG_LIBDIR=/usr/ports/cad/surelog/work/.pkgconfig:/usr/local/libdata/pkgconfig:/usr/local/share/pkgconfig:/usr/libdata/pkgconfig SHELL=/bin/sh CONFIG_SHELL=/bin/sh CMAKE_PREFIX_PATH="/usr/local" CCACHE_DIR="/tmp/.ccache" /usr/local/bin/cmake -DFREEBSD_JAVA_VERSION=17  -DPython3_EXECUTABLE=/usr/local/bin/python3.9 -DSURELOG_WITH_PYTHON:BOOL=false -DCMAKE_C_COMPILER:STRING="cc"  -DCMAKE_CXX_COMPILER:STRING="c++"  -DCMAKE_C_FLAGS:STRING="-O2 -pipe  -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing "  -DCMAKE_C_FLAGS_DEBUG:STRING="-O2 -pipe  -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing "  -DCMAKE_C_FLAGS_RELEASE:STRING="-O2 -pipe  -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing  -DNDEBUG"  -DCMAKE_CXX_FLAGS:STRING="-O2 -pipe -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing  -isystem /usr/local/include "  -DCMAKE_CXX_FLAGS_DEBUG:STRING="-O2 -pipe -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing  -isystem /usr/local/include "  -DCMAKE_CXX_FLAGS_RELEASE:STRING="-O2 -pipe -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing  -isystem /usr/local/include  -DNDEBUG"  -DCMAKE_EXE_LINKER_FLAGS:STRING=" -fstack-protector-strong -L/usr/local/lib "  -DCMAKE_MODULE_LINKER_FLAGS:STRING=" -fstack-protector-strong -L/usr/local/lib "  -DCMAKE_SHARED_LINKER_FLAGS:STRING=" -fstack-protector-strong -L/usr/local/lib "  -DCMAKE_INSTALL_PREFIX:PATH="/usr/local"  -DCMAKE_BUILD_TYPE:STRING="Release"  -DTHREADS_HAVE_PTHREAD_ARG:BOOL=YES  -DCMAKE_INSTALL_RPATH_USE_LINK_PATH=YES  -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON  -DFETCHCONTENT_FULLY_DISCONNECTED:BOOL=ON -DBUILD_SHARED_LIBS:BOOL=ON -DSURELOG_BUILD_TESTS:BOOL=OFF -GNinja -DPython_ADDITIONAL_VERSIONS=3.9 -DBOOST_PYTHON_SUFFIX:STRING=39 /usr/ports/cad/surelog/work/Surelog-1.50
-- The C compiler identification is Clang 14.0.5
-- The CXX compiler identification is Clang 14.0.5
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/libexec/ccache/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/libexec/ccache/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Building version v1.50 [release]
-- Proceeding with version: 23.3.3.0
-- Looking for strtof_l
-- Looking for strtof_l - not found
-- Looking for strtoull_l
-- Looking for strtoull_l - not found
-- Looking for realpath
-- Looking for realpath - found
-- CMAKE_CXX_FLAGS: -O2 -pipe -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing  -isystem /usr/local/include
-- Building without demo. To enable demo build use: -DWITH_DEMO=True
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Python3: /usr/local/bin/python3.9 (found version "3.9.16") found components: Interpreter 
-- Python3_EXECUTABLE = /usr/local/bin/python3.9
-- Looking for C++ include initializer_list
-- Looking for C++ include initializer_list - found
-- Found ZLIB: /usr/lib/libz.so (found version "1.2.12") 
-- Found Python: /usr/local/bin/python3.9 (found version "3.9.16") found components: Interpreter 
-- Found Python3: /usr/local/bin/python3.9 (found suitable version "3.9.16", minimum required is "3.3") found components: Interpreter 
-- Python3_EXECUTABLE = /usr/local/bin/python3.9
CMake Warning at CMakeLists.txt:151 (message):
  Using tcmalloc; if this creates runtime issues, configure with
  -DSURELOG_WITH_TCMALLOC=Off

-- Found tcmalloc: /usr/local/lib/libtcmalloc.so
-- Found Java: /usr/local/bin/java (found suitable version "17.0.6", minimum required is "17") found components: Runtime 
-- Java_JAVA_EXECUTABLE = /usr/local/bin/java
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    BOOST_PYTHON_SUFFIX
    CMAKE_MODULE_LINKER_FLAGS
    CMAKE_VERBOSE_MAKEFILE
    FETCHCONTENT_FULLY_DISCONNECTED
    Python_ADDITIONAL_VERSIONS

-- Build files have been written to: /usr/ports/cad/surelog/work/.build
```

hzeller, 2023-03-30:
This sounds probably something where maybe @timkpaine might have a hunch what could be problematic here.

The pkg-config changes have been fairly recent, but I think it should install. There have been a few little changes on the pkg-config files since 1.50, but probably nothing that would fix your particular issue here.
But might be worthwhile to `git pull ; git submodule update --init --recursive`

timkpaine, 2023-03-30:
That's quite a large set of cmake variable overrides. I would start by replicating what CI does but without installing, and add back variables until it breaks. In particular you're overriding pkg-config variables and data files variables which might interact poorly with recent changes.

hzeller, 2023-04-26:
Did this resolve ? From all my experiments, that should work. All the work on pkg-config and cmake outputs that @timkpaine was contributing is settled now and works well.

yurivict, 2023-04-26:
I switched the FreeBSD port to ude external UHDM, so I don't test this any more.

hzeller, 2023-04-26:
Ok, closing then.

closed 2023-04-26 by hzeller