chipsalliance/Surelog #3563: 'make install' runs long 'surelog' processes instead of just installing files

yurivict, 2023-04-05:
'make build' runs long surelog processes in the end. 'make install' runs long surelog processes again. All files should be built in 'make build', not in 'make install'. I suspect that 'make install' reruns the same surelog runs that 'make build' runs.

yurivict, 2023-04-05:
@timkpaine 

> There is not enough information here to know what you're asking.

```make build``` is run first. It builds the package.
```make install``` is run second. It installs the package.

Do you know why ```make install``` runs surelog commands instead of installing the package?

timkpaine, 2023-04-05:
`make build` doesn't appear anywhere in the install documentation: https://github.com/chipsalliance/Surelog/blob/master/INSTALL.md

My recommendation is to follow the documentation. Presumably you're using cmake to generate makefiles to build stuff. But you'd need to provide the commands you're using and logs to know what specific issues you're running into.