chipsalliance/Surelog #3582: 1.54: error: duplicate case value: 'uhdmregs' and 'uhdmvoid_typespec' both equal '2247'

yurivict, 2023-04-11:
```
/usr/ports/cad/uhdm/work/.build/generated/src/Serializer.cpp:436:8: error: duplicate case value: 'uhdmregs' and 'uhdmvoid_typespec' both equal '2247'
  case uhdmvoid_typespec /* = 2247 */: return "void_typespec";
       ^
/usr/ports/cad/uhdm/work/.build/generated/src/Serializer.cpp:307:8: note: previous case defined here
  case uhdmregs /* = 2247 */: return "regs";
       ^
```

case values should all be different.

uhdm-1.54

hzeller, 2023-04-11:
This sounds weird. I just tried a fresh checkout to version 1.54; after a make the uhdm_types.h shows different forms:

```
$ git checkout v1.54
$ make clean
$ make
$ find build -type f | xargs egrep "(uhdmvoid_typespec|uhdmregs).*[0-9]+"
build/generated/src/Serializer.cpp:  case uhdmregs /* = 2247 */: return "regs";
build/generated/src/Serializer.cpp:  case uhdmvoid_typespec /* = 2376 */: return "void_typespec";
build/generated/src/Serializer.cpp:  case uhdmvoid_typespec /* = 2376 */: return void_typespecMaker.objects_[index];
build/generated/uhdm/uhdm_types.h:  uhdmregs = 2247,
build/generated/uhdm/uhdm_types.h:  uhdmvoid_typespec = 2376,
```

Maybe there is a confusion as the headers are included as system headers (with angle brackets) and thus things break in unexpected ways if there is already a `<uhdm/...>` on the system ?

If so, this should probably  be properly worked around in CMakeLists.

yurivict, 2023-04-11:
> Maybe there is a confusion as the headers [...]

I only have one set of UGDM headers installed into /usr/local/include/UHDM. There is no UHDM git submodule.

alaindargelas, 2023-04-12:
@yurivict it looks like you have incremental build issues, see:
https://github.com/chipsalliance/Surelog/issues/3575

I can't reproduce this error either. Please do a clean build.

closed 2023-04-13 by alaindargelas