chipsalliance/Surelog #3596: surelog installs empty directories

yurivict, 2023-04-16:
It installs this empty directory hierarchy:
```
/usr/local/share/surelog
/usr/local/share/surelog/pkg
/usr/local/share/surelog/pkg/work
```

It doesn't make sense to install empty directories into a read-only location.

timkpaine, 2023-04-16:
`ovm_pkg.sv.slpa  ovm_pkg.sv.slpp  uvm_pkg.sv.slpa  uvm_pkg.sv.slpp` are in that folder

closed 2023-04-16 by timkpaine

yurivict, 2023-04-16:
@timkpaine 

cmake didn't install anything into that folder.
What might be wrong?

timkpaine, 2023-04-16:
@yurivict you really need to get into the habit of posting configuration, commands, and logs. This is not the first time you've failed to do so, and it's difficult to provide any help without them. This is a good guide, in particular the repro section: https://stackoverflow.com/help/how-to-ask

yurivict, 2023-04-16:
I don't provide details because I don;t build by hand typing in commands.
The build is done by standard cmake commands that are used for thousands of other packages.
It means that in default configuration under common conditions these directories are left empty.

[Here](https://people.freebsd.org/~yuri/surelog-1.56.log) is a complete build log showing that these directories are installed empty.

timkpaine, 2023-04-16:
I can tell from the logs that this is untrue, just look at all the manually specified cmake variables. We test heavily across lots of different configurations and operating systems, if you're not comfortable debugging issues yourself the exact sequence of cmake commands, environment variable overrides, and other information about the system is essential.

yurivict, 2023-04-16:
They are not manually specified - they are set like this by the ports framework. When qBittorrent is built - the config line is largely the same.

timkpaine, 2023-04-16:
These lines build the precompiled assets: https://github.com/chipsalliance/Surelog/blob/master/CMakeLists.txt#L721
And they get installed a bit later: https://github.com/chipsalliance/Surelog/blob/master/CMakeLists.txt#L815

Your logs show the assets being built, but they maybe they're not ending up in the right output folder. We set the var here: https://github.com/chipsalliance/Surelog/blob/master/CMakeLists.txt#L140

You can see the correct installation in CI: https://github.com/chipsalliance/Surelog/actions/runs/4710503128/jobs/8354210283#step:7:240

I would try to build outside of any framework which might set lots of different CMake vars, and see if that yields a different result. It's possible that we rely on a variable without setting it. You could also try removing these redirects and/or checking the build logs to see if they ended up in a different folder: https://github.com/chipsalliance/Surelog/blob/master/CMakeLists.txt#L731

yurivict, 2023-04-16:
BTW, did you test build with a stage directory? My builds use stage directory.

timkpaine, 2023-04-16:
No, I would normally only use that for `makefile`-driven projects, we just use makefiles as a scripting layer, so I wouldn't really expect that to work. I think [PREFIX](https://github.com/chipsalliance/Surelog/blob/master/Makefile#L23)) dictates the path so it would be `make build install PREFIX=/tmp/surelog-stagedir`

yurivict, 2023-04-16:
No, all packaging systems use stage directories because they don't need to install into $PREFIX in order to create a package.

stage directories work only at the level of 'make', not 'cmake', with the above command.
Some projects are broken when stage directory is used. It looks like this might be the case for Surelog.

yurivict, 2023-04-16:
```CMAKE_INSTALL_PREFIX``` is just the ```$PREFIX``` variable value. stage directory is prepended to it by make.

yurivict, 2023-04-16:
In my build directory ```work/.build``` ```work/.build/bin/pkg``` is empty.

timkpaine, 2023-04-16:
> stage directories work only at the level of 'make', not 'cmake', with the above command. Some projects are broken when stage directory is used. It looks like this might be the case for Surelog.

We don't have any dependency on `make`, I would avoid trying to mix `make` and `cmake` when you build and install.

yurivict, 2023-04-16:
Either make or ninja are always run as a backend of cmake.

timkpaine, 2023-04-16:
> Either make or ninja are always run as a backend of cmake.

Or e.g. visual studio, but the point stands. The top level makefile != the makefiles generated by cmake and shouldn't be treated as such.