chipsalliance/Surelog #3597: The part_select operator isn't linked with the variable from which it selects bits

yurivict, 2023-04-16:
I am looking at these 2 testcases:

This case:
```
module top (clk, in, out);
        input wire [0:0] clk;
        input wire [7:0] in;
        output wire [7:0] out;
        output reg [7:0] out;

        assign out = in[(4'h3):(3'h1)];
endmodule
```
has part_select that references 'in' (through _ref_obj):
```
    \_part_select: , line:7:15, endln:7:32
      |vpiParent:
      \_ref_obj: in (work@top.in), line:7:15, endln:7:17
        |vpiParent:
        \_cont_assign: , line:7:9, endln:7:32
        |vpiName:in
        |vpiFullName:work@top.in
        |vpiDefName:in
      |vpiConstantSelect:1
      |vpiLeftRange:
      \_constant: , line:7:19, endln:7:23
        |vpiDecompile:4'h3
        |vpiSize:4
        |HEX:3
        |vpiConstType:5
      |vpiRightRange:
      \_constant: , line:7:26, endln:7:30
        |vpiDecompile:3'h1
        |vpiSize:3
        |HEX:1
        |vpiConstType:5
```

This case:
```
module top (clk, in, out);
        input wire [0:0] clk;
        input wire [7:0] in;
        output wire [7:0] out;
        output reg [7:0] out;

        always @(posedge clk)
                if (in[(4'h3):(3'h1)])
                        out <= in;
endmodule
```
has part_select that ***DOES NOT*** reference 'in' at all:
```
        \_part_select: , line:8:7, endln:8:24
          |vpiParent:
          \_if_stmt: , line:8:3, endln:9:14
          |vpiConstantSelect:1
          |vpiLeftRange:
          \_constant: , line:8:11, endln:8:15
            |vpiDecompile:4'h3
            |vpiSize:4
            |HEX:3
            |vpiConstType:5
          |vpiRightRange:
          \_constant: , line:8:18, endln:8:22
            |vpiDecompile:3'h1
            |vpiSize:3
            |HEX:1
            |vpiConstType:5
```

Please note that in the first case part_select has a parent vpiRhs:
```
  \_cont_assign: , line:7:9, endln:7:32
    |vpiParent:
    \_module_inst: work@top (work@top), file:/usr/home/yuri/issue-part-select/assign+part-select.sv, line:1:1, endln:8:10
    |vpiRhs:
    \_part_select: , line:7:15, endln:7:32
      |vpiParent:
      \_ref_obj: in (work@top.in), line:7:15, endln:7:17
```
but the vpiParent element under _part_select says that the parent is _ref_obj. Does this make any sense? This looks strange to me.

Shouldn't _part_select have another argument for the wire that it selects the value from instead it being set as a parent?

Version: 1.56
FreeBSD 13.1

alaindargelas, 2023-04-16:
We use ref_obj everywhere in UHDM to defer binding. It is a means to attach an object immediately that holds a name while compiling a statement, then later on we can bind that name to an actual object.

There might be a bug here:
module top (clk, in, out);
        input wire [0:0] clk;
        input wire [7:0] in;
        output wire [7:0] out;
        output reg [7:0] out;

        always @(posedge clk)
                if (in[(4'h3):(3'h1)])
                        out <= in;
endmodule

alaindargelas, 2023-05-01:
Fixed by #3628

closed 2023-05-01 by alaindargelas