chipsalliance/Surelog #3598: The same net's definition is listed twice

yurivict, 2023-04-17:
This module:
```
module top (out);
        output wire [7:0] out;

        reg signed [7:0] r = (1'h0);

        assign out = r;
endmodule
```

Produces this Module that has the "r" net listed twice (see the ```------------------``` lines):
```
|uhdmtopModules:
\_module_inst: work@top (work@top), file:/usr/home/yuri/nn-trainer/verilog-parser/reg-const.sv, line:1:1, endln:7:10
  |vpiName:work@top
  |vpiDefName:work@top
  |vpiTop:1
  |vpiNet:
---------------------------------- the first time ----------------------------------
  \_logic_net: (work@top.r), line:4:19, endln:4:29
    |vpiParent:
    \_module_inst: work@top (work@top), file:/usr/home/yuri/nn-trainer/verilog-parser/reg-const.sv, line:1:1, endln:7:10
    |vpiTypespec:
    \_logic_typespec: , line:4:2, endln:4:18
      |vpiRange:
      \_range: , line:4:13, endln:4:18
        |vpiLeftRange:
        \_constant: , line:4:14, endln:4:15
          |vpiParent:
          \_range: , line:4:13, endln:4:18
          |vpiDecompile:7
          |vpiSize:64
          |UINT:7
          |vpiConstType:9
        |vpiRightRange:
        \_constant: , line:4:16, endln:4:17
          |vpiParent:
          \_range: , line:4:13, endln:4:18
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
          |vpiConstType:9
      |vpiSigned:1
    |vpiName:r
    |vpiFullName:work@top.r
    |vpiNetType:48
    |vpiSigned:1
  |vpiNet:
  \_logic_net: (work@top.out), line:1:13, endln:1:16
    |vpiParent:
    \_module_inst: work@top (work@top), file:/usr/home/yuri/nn-trainer/verilog-parser/reg-const.sv, line:1:1, endln:7:10
    |vpiTypespec:
    \_logic_typespec: , line:2:9, endln:2:19
      |vpiRange:
      \_range: , line:2:14, endln:2:19
        |vpiLeftRange:
        \_constant: , line:2:15, endln:2:16
          |vpiParent:
          \_range: , line:2:14, endln:2:19
          |vpiDecompile:7
          |vpiSize:64
          |UINT:7
          |vpiConstType:9
        |vpiRightRange:
        \_constant: , line:2:17, endln:2:18
          |vpiParent:
          \_range: , line:2:14, endln:2:19
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
          |vpiConstType:9
    |vpiName:out
    |vpiFullName:work@top.out
    |vpiNetType:1
  |vpiTopModule:1
  |vpiPort:
  \_port: (out), line:1:13, endln:1:16
    |vpiParent:
    \_module_inst: work@top (work@top), file:/usr/home/yuri/nn-trainer/verilog-parser/reg-const.sv, line:1:1, endln:7:10
    |vpiName:out
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: (work@top.out), line:1:13, endln:1:16
      |vpiName:out
      |vpiFullName:work@top.out
      |vpiActual:
      \_logic_net: (work@top.out), line:1:13, endln:1:16
    |vpiTypedef:
    \_logic_typespec: , line:2:9, endln:2:19
      |vpiRange:
      \_range: , line:2:14, endln:2:19
        |vpiParent:
        \_port: (out), line:1:13, endln:1:16
        |vpiLeftRange:
        \_constant: , line:2:15, endln:2:16
          |vpiParent:
          \_range: , line:2:14, endln:2:19
          |vpiDecompile:7
          |vpiSize:64
          |UINT:7
          |vpiConstType:9
        |vpiRightRange:
        \_constant: , line:2:17, endln:2:18
          |vpiParent:
          \_range: , line:2:14, endln:2:19
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
          |vpiConstType:9
  |vpiContAssign:
  \_cont_assign: , line:4:19, endln:4:20
    |vpiNetDeclAssign:1
    |vpiRhs:
    \_constant: , line:4:24, endln:4:28
      |vpiParent:
      \_cont_assign: , line:4:19, endln:4:20
      |vpiDecompile:1'h0
      |vpiSize:8
      |HEX:0
      |vpiConstType:5
    |vpiLhs:
---------------------------------- the second time ----------------------------------
    \_logic_net: (r), line:4:19, endln:4:29
      |vpiParent:
      \_cont_assign: , line:4:19, endln:4:20
      |vpiTypespec:
      \_logic_typespec: , line:4:2, endln:4:18
        |vpiParent:
        \_logic_net: (r), line:4:19, endln:4:29
        |vpiRange:
        \_range: , line:4:13, endln:4:18
          |vpiParent:
          \_logic_typespec: , line:4:2, endln:4:18
          |vpiLeftRange:
          \_constant: , line:4:14, endln:4:15
            |vpiParent:
            \_range: , line:4:13, endln:4:18
            |vpiDecompile:7
            |vpiSize:64
            |UINT:7
            |vpiConstType:9
          |vpiRightRange:
          \_constant: , line:4:16, endln:4:17
            |vpiParent:
            \_range: , line:4:13, endln:4:18
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
            |vpiConstType:9
        |vpiSigned:1
      |vpiName:r
      |vpiNetType:48
      |vpiSigned:1
  |vpiContAssign:
  \_cont_assign: , line:6:9, endln:6:16
```

It looks like ref_obj should be used under the Assign operator.

Version: 1.56
clang-15
FreeBSD 13.2

alaindargelas, 2023-04-18:
Expressions are uniquified (Cloned) for the same reason you have to traverse all the instances to make sure you don't alias objects that should not be aliased. The cloning is brute force hence the "duplication"

closed 2023-04-18 by alaindargelas

yurivict, 2023-04-18:
But in this case it seems like it should be aliased. These 2 is the same net. As long as the name is the same - this is the same net.

alaindargelas, 2023-04-18:
There is no aliasing mechanism in the cloning. As said above it is brute force.
It is not a problem for most applications, including Synthesis. What is the problem you face?
You can create your own symbol table.