chipsalliance/Surelog #3602: Support flattening of array with assignment pattern inside another assignment pattern

kamilrakoczy, 2023-04-17:
TEST:
```
package ctrl_pkg;
  parameter int RegNumBanks = 2;
  parameter int RegPagesPerBank = 64;
endpackage

package pkg;
  parameter int unsigned NumBanks        = ctrl_pkg::RegNumBanks;
  parameter int unsigned PagesPerBank    = ctrl_pkg::RegPagesPerBank;
  parameter int HwDataRules = 2;
  parameter int MuBi4Width = 4;
  typedef enum logic [MuBi4Width-1:0] {
    MuBi4True = 4'h6,
    MuBi4False = 4'h9 
  } mubi4_t;
  typedef enum logic [1:0] {
    PhaseSeed,
    PhaseRma,
    PhaseNone,
    PhaseInvalid
  } flash_lcmgr_phase_e;
  typedef struct packed {
    mubi4_t en;
  } mp_region_cfg_t;
  typedef struct packed {
    flash_lcmgr_phase_e   phase;
    mp_region_cfg_t cfg;
  } data_region_attr_t;
  parameter data_region_attr_t HwDataAttr[HwDataRules] = '{
    '{
       phase: PhaseSeed,
       cfg:   '{ en:          MuBi4True }
     },
    '{
       phase: PhaseSeed,
       cfg:   '{ en:          MuBi4True }
     }
  };

endpackage

module foo
import pkg::*;
(
  input data_region_attr_t DATA_REG);
endmodule
module top();
import pkg::*;
foo f(.DATA_REG(HwDataAttr));

endmodule
```
Similar test to one in: https://github.com/chipsalliance/Surelog/pull/3601, but this time we have array that contains assignment pattern inside another assignment pattern. Surelog currently doesn't flatten such assignment patterns and it would be good if they were flattened. (Opentitan contains both cases)

alaindargelas, 2023-04-28:
@kamilrakoczy, your testcase is incorrect, it does not compile in any simulator.
I think you meant, where the data types of the parameters match and you actually override a parameter and not assign a constant to a port, correct? Can you check the original source code in Opentitan or point me to the location where this testcase was coming from?
Similarly, the testcase you added in #3601 contains the same error, you are assigning a port to a parameter value, where I guess it needs to be a parameter override. If that is the case, please fix the checked in testcase in #3601. You'll notice if you do that, that the pattern assign gets flattened even more to constants, because Surelog performs more constant pushing on parameters.

```

package ctrl_pkg;
  parameter int RegNumBanks = 2;
  parameter int RegPagesPerBank = 64;
endpackage

package pkg;
  parameter int unsigned NumBanks        = ctrl_pkg::RegNumBanks;
  parameter int unsigned PagesPerBank    = ctrl_pkg::RegPagesPerBank;
  parameter int HwDataRules = 2;
  parameter int MuBi4Width = 4;
  typedef enum logic [MuBi4Width-1:0] {
    MuBi4True = 4'h6,
    MuBi4False = 4'h9 
  } mubi4_t;
  typedef enum logic [1:0] {
    PhaseSeed,
    PhaseRma,
    PhaseNone,
    PhaseInvalid
  } flash_lcmgr_phase_e;
  typedef struct packed {
    mubi4_t en;
  } mp_region_cfg_t;
  typedef struct packed {
    flash_lcmgr_phase_e   phase;
    mp_region_cfg_t cfg;
  } data_region_attr_t;
  parameter data_region_attr_t HwDataAttr[HwDataRules] = '{
    '{
       phase: PhaseSeed,
       cfg:   '{ en:          MuBi4True }
     },
    '{
       phase: PhaseSeed,
       cfg:   '{ en:          MuBi4True }
     }
  };

endpackage

module foo
import pkg::*;
  #(data_region_attr_t DATA_REG[HwDataRules])();
endmodule

module top();
import pkg::*;
  foo #(.DATA_REG(HwDataAttr)) f ();

endmodule
```

kamilrakoczy, 2023-04-28:
I've took original test case from:
- top module: https://github.com/lowRISC/opentitan/blob/master/hw/ip/flash_ctrl/rtl/flash_mp.sv#L168
- foo module: https://github.com/lowRISC/opentitan/blob/master/hw/ip/flash_ctrl/rtl/flash_mp_data_region_sel.sv#L16
- pkg              : https://github.com/lowRISC/opentitan/blob/master/hw/ip/flash_ctrl/rtl/flash_ctrl_pkg.sv#L293

So we want to assign constant to the port, but now I see that I forgot to add ``[HwDataRules]`` to ``input`` in ``foo`` when I adapted test case from #3601. In opentitan there is also only single element in ``HwDataAttr`` array, but I've extended it to two elements (if it is much harder for two elements, I think we can stick to single element for now, as I think there isn't case in opentitan with more than one element). Updated test case with added array:

```
package ctrl_pkg;
  parameter int RegNumBanks = 2;
  parameter int RegPagesPerBank = 64;
endpackage

package pkg;
  parameter int unsigned NumBanks        = ctrl_pkg::RegNumBanks;
  parameter int unsigned PagesPerBank    = ctrl_pkg::RegPagesPerBank;
  parameter int HwDataRules = 2;
  parameter int MuBi4Width = 4;
  typedef enum logic [MuBi4Width-1:0] {
    MuBi4True = 4'h6,
    MuBi4False = 4'h9 
  } mubi4_t;
  typedef enum logic [1:0] {
    PhaseSeed,
    PhaseRma,
    PhaseNone,
    PhaseInvalid
  } flash_lcmgr_phase_e;
  typedef struct packed {
    mubi4_t en;
  } mp_region_cfg_t;
  typedef struct packed {
    flash_lcmgr_phase_e   phase;
    mp_region_cfg_t cfg;
  } data_region_attr_t;
  parameter data_region_attr_t HwDataAttr[HwDataRules] = '{
    '{
       phase: PhaseSeed,
       cfg:   '{ en:          MuBi4True }
     },
    '{
       phase: PhaseSeed,
       cfg:   '{ en:          MuBi4True }
     }
  };

endpackage

module foo
import pkg::*;
(
  input data_region_attr_t DATA_REG[HwDataRules]);
endmodule
module top();
import pkg::*;
foo f(.DATA_REG(HwDataAttr));

endmodule
```

alaindargelas, 2023-04-30:
#3627 performs the flattening for one element for now.

closed 2023-04-30 by alaindargelas