chipsalliance/Surelog #3607: Missing way to get original ``struct typespec`` from ``string typespec``

kamilrakoczy, 2023-04-20:
TEST:
```
module ibex_csr #(
parameter int Width = 32,
parameter bit [Width-1:0] ResetValue = '0
)
();

endmodule
module dut();
typedef enum logic [1:0] {
  PMP_MODE_OFF   = 2'b00,
  PMP_MODE_TOR   = 2'b01,
  PMP_MODE_NA4   = 2'b10,
  PMP_MODE_NAPOT = 2'b11
} pmp_cfg_mode_e;
typedef struct packed {
  logic          lock;
  pmp_cfg_mode_e mode;
  logic          exec;
  logic          write;
  logic          read;
} pmp_cfg_t;

localparam pmp_cfg_t pmp_cfg_rst[1] = '{
  '{lock: 1'b0, mode: PMP_MODE_OFF, exec: 1'b0, write: 1'b0, read: 1'b0}
};

  ibex_csr #(
    .Width($bits(pmp_cfg_t)),
    .ResetValue(pmp_cfg_rst[0])
  ) u_pmp_cfg_csr ();
endmodule
```
UHDM:
```
|uhdmtopModules:
...
     |vpiParamAssign:
     \_param_assign: , line:3:27, endln:3:42
       |vpiParent:
       \_module_inst: work@ibex_csr (work@dut.u_pmp_cfg_csr), file:dut.sv, line:27:3, endln:30:22
       |vpiOverriden:1
       |vpiRhs:
       \_operation: , line:24:3, endln:24:73
         |vpiParent:
         \_param_assign: , line:3:27, endln:3:42
         |vpiSize:6
         |vpiOpType:75
         |vpiOperand:
         \_tagged_pattern: , line:24:11, endln:24:15
           |vpiParent:
           \_operation: , line:24:3, endln:24:73
           |vpiPattern:
           \_constant: , line:24:11, endln:24:15
             |vpiParent:
             \_tagged_pattern: , line:24:11, endln:24:15
             |vpiDecompile:1'b0
             |vpiSize:1
             |BIN:0
             |vpiConstType:3
           |vpiTypespec:
           \_string_typespec: (lock), line:24:5, endln:24:9
             |vpiParent:
             \_tagged_pattern: , line:24:11, endln:24:15
             |vpiName:lock
         |vpiOperand:
         \_tagged_pattern: , line:24:23, endln:24:35
           |vpiParent:
           \_operation: , line:24:3, endln:24:73
           |vpiPattern:
           \_constant: , line:24:23, endln:24:35
             |vpiParent:
             \_tagged_pattern: , line:24:23, endln:24:35
             |vpiDecompile:2'b00
             |vpiSize:2
             |BIN:00
             |vpiConstType:3
           |vpiTypespec:
           \_string_typespec: (mode), line:24:17, endln:24:21
             |vpiParent:
             \_tagged_pattern: , line:24:23, endln:24:35
             |vpiName:mode
         |vpiOperand:
         \_tagged_pattern: , line:24:43, endln:24:47
           |vpiParent:
           \_operation: , line:24:3, endln:24:73
           |vpiPattern:
           \_constant: , line:24:43, endln:24:47
             |vpiParent:
             \_tagged_pattern: , line:24:43, endln:24:47
             |vpiDecompile:1'b0
             |vpiSize:1
             |BIN:0
             |vpiConstType:3
           |vpiTypespec:
           \_string_typespec: (exec), line:24:37, endln:24:41
             |vpiParent:
             \_tagged_pattern: , line:24:43, endln:24:47
             |vpiName:exec
          |vpiOperand:
         \_tagged_pattern: , line:24:56, endln:24:60
           |vpiParent:
           \_operation: , line:24:3, endln:24:73
           |vpiPattern:
           \_constant: , line:24:56, endln:24:60
             |vpiParent:
             \_tagged_pattern: , line:24:56, endln:24:60
             |vpiDecompile:1'b0
             |vpiSize:1
             |BIN:0
             |vpiConstType:3
           |vpiTypespec:
           \_string_typespec: (write), line:24:49, endln:24:54
             |vpiParent:
             \_tagged_pattern: , line:24:56, endln:24:60
             |vpiName:write
         |vpiOperand:
         \_tagged_pattern: , line:24:68, endln:24:72
           |vpiParent:
           \_operation: , line:24:3, endln:24:73
           |vpiPattern:
           \_constant: , line:24:68, endln:24:72
             |vpiParent:
             \_tagged_pattern: , line:24:68, endln:24:72
             |vpiDecompile:1'b0
             |vpiSize:1
             |BIN:0
             |vpiConstType:3
           |vpiTypespec:
           \_string_typespec: (read), line:24:62, endln:24:66
             |vpiParent:
             \_tagged_pattern: , line:24:68, endln:24:72
             |vpiName:read
       |vpiLhs:
       \_parameter: (work@dut.u_pmp_cfg_csr.ResetValue), line:3:27, endln:3:37
```
When module parameter is set to assignment pattern (but this parameter isn't custom type), in UHDM there is ``assignment pattern op`` to wire that doesn't have ``struct typespec``. To workaround this, I could use ``struct typespec`` that each ``string typespec`` is from, but I don't think Surelog allows this.

Is there a way to get original ``struct typespec`` from ``string typespec`` that defines struct field?

alaindargelas, 2023-04-26:
The typespec of the operation has been lost during the parameter overloading. If you notice at the top level, the operation has a payload describing the struct:
```
    \_operation: , id:148, line:24:39, endln:26:2
      |vpiParent:
      \_param_assign: , id:99, line:24:22, endln:26:2
      |vpiTypespec:
      \_array_typespec: , id:149
        |vpiParent:
        \_operation: , id:148, line:24:39, endln:26:2
        |vpiRange:
        \_range: , id:150, line:24:34, endln:24:35
          |vpiParent:
          \_array_typespec: , id:149
          |vpiLeftRange:
          \_constant: , id:151, line:24:34, endln:24:35
            |vpiParent:
            \_range: , id:150, line:24:34, endln:24:35
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
            |vpiConstType:9
          |vpiRightRange:
          \_operation: , id:152
            |vpiParent:
            \_range: , id:150, line:24:34, endln:24:35
            |vpiOpType:11
            |vpiOperand:
            \_constant: , id:153, line:24:34, endln:24:35
              |vpiParent:
              \_operation: , id:152
              |vpiDecompile:1
              |vpiSize:64
              |UINT:1
              |vpiConstType:9
            |vpiOperand:
            \_constant: , id:154
              |vpiParent:
              \_operation: , id:152
              |vpiSize:64
              |INT:1
              |vpiConstType:7
        |vpiElemTypespec:
        \_struct_typespec: (pmp_cfg_t), id:155, line:16:9, endln:16:15
          |vpiParent:
          \_array_typespec: , id:149
          |vpiName:pmp_cfg_t
          |vpiPacked:1
          |vpiTypespecMember:
          \_typespec_member: (lock), id:156, line:17:18, endln:17:22
            |vpiParent:
            \_struct_typespec: (pmp_cfg_t), id:155, line:16:9, endln:16:15
            |vpiName:lock
            |vpiTypespec:
            \_logic_typespec: , id:157, line:17:3, endln:17:8
              |vpiParent:
              \_typespec_member: (lock), id:156, line:17:18, endln:17:22
```

alaindargelas, 2023-04-26:
I'll have to understand how did the payload (operator typespec) has been lost.

alaindargelas, 2023-04-27:
Fixed by #3619, you can get the operation typespec.

closed 2023-04-27 by alaindargelas