chipsalliance/Surelog #3617: How do you write .sv from UHDM?

maxslug, 2023-04-25:
I can use surelog to read in .sv and elaborate it, then use the UHDM python VPI / SWIG interface to interrogate it, yay!

Sorry if I'm missing something obvious, but is there a utiltiy to create a .sv (eg. verilog 2017) from the UHDM?
Follow-up: for elaborated and non-elaborated?

Thanks!

Thomasb81, 2023-04-25:
You can have a look here : https://github.com/chipsalliance/Surelog/blob/master/src/roundtrip.cpp

mithro, 2023-04-25:
FWIW - You can also import the UHDM into Yosys (https://github.com/chipsalliance/yosys-f4pga-plugins/tree/main/systemverilog-plugin) and then use Yosys to write out verilog (https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/write_verilog.html).

alaindargelas, 2023-04-26:
@mithro's suggestion is going to write out Verilog 2005. 
@Thomasb81's suggestion to use @hs-apotell's rountrip executable shows the general idea to traverse UHDM but does not get you quite there.

@maxslug There is an embryonic pretty printer here that prints SV expressions from UHDM: https://github.com/chipsalliance/UHDM/blob/3852e15864f362dca23b3c20456e77f3254e41c5/templates/ExprEval.cpp#L746
"Someone" has to pick up from there and write the code to pretty print the entire design.
It could be done in Python too if you feel like it!

closed 2023-05-27 by alaindargelas