chipsalliance/Surelog #3630: Unsupported typespec when file level typedef uses package type

kamilrakoczy, 2023-05-04:
Originally reported here: https://github.com/chipsalliance/yosys-f4pga-plugins/issues/502
TEST:
```
package foo_flags;
  typedef struct packed {
    logic a;
    logic b;
    logic c;
  } common_flags_t;
endpackage : foo_flags

package fooes;

  typedef enum logic [1:0]{
    a,
    b,
    c,
    d
  } classes;

endpackage : fooes

typedef struct packed {
  logic a;
  fooes::classes b;
} padded_fooes_t;

typedef union packed {
  foo_flags::common_flags_t  [3:0][7:0] atype_t;
  padded_fooes_t     [3:0][7:0] btype_t;
} top_flag_t;

module top(input top_flag_t a , output top_flag_t b);
  assign b = a;
endmodule
```

UHDM:
```
    |vpiPort: 
    \_port: (a), line:31:29, endln:31:30
      |vpiParent:
      \_module_inst: work@top (work@top), file:dut.sv, line:31:1, endln:33:10
      |vpiName:a
      |vpiDirection:1
      |vpiLowConn:
      \_ref_obj: 
        |vpiActual:
        \_logic_net: (work@top.a), line:31:29, endln:31:30
      |vpiTypedef:
      \_union_typespec: (top_flag_t), line:26:9, endln:26:14
        |vpiParent:
        \_design: (work@top)
        |vpiName:top_flag_t
        |vpiInstance:
        \_design: (work@top)
        |vpiPacked:1
        |vpiTypespecMember:
        \_typespec_member: (atype_t), line:27:41, endln:27:48
          |vpiParent:
          \_union_typespec: (top_flag_t), line:26:9, endln:26:14
          |vpiName:atype_t
          |vpiTypespec:
          \_unsupported_typespec: (foo_flags::common_flags_t), line:27:3, endln:27:14
            |vpiParent:
            \_typespec_member: (atype_t), line:27:41, endln:27:48
            |vpiName:foo_flags::common_flags_t
          |vpiRefFile:dut.sv
          |vpiRefLineNo:27
          |vpiRefColumnNo:3
          |vpiRefEndLineNo:27
          |vpiRefEndColumnNo:40
```

When typedef is defined on file level and it is using some package type, Surelog reports  this package type as ``_unsupported_typespec``.

QuantamHD, 2023-05-06:
@alaindargelas this one is important for a Google design

alaindargelas, 2023-05-06:
Fixed by #3634

closed 2023-05-06 by alaindargelas