chipsalliance/Surelog #3637: Exception while parsing Opentitan after #3632

kamilrakoczy, 2023-05-08:
After #3632 Surelog throws exception while parsing Opentitan:
```
terminate called after throwing an instance of 'std::length_error'
  what():  basic_string::_M_create
```
Backtrace:
```
(gdb) bt                                                  
#0  0x00007ffff769f26c in ?? () from /usr/lib/libc.so.6                                                                                                                                                                                                                                                                       
#1  0x00007ffff764fa08 in raise () from /usr/lib/libc.so.6                                                                                                                                                                                                                                                                    
#2  0x00007ffff7638538 in abort () from /usr/lib/libc.so.6                                                                                                     
#3  0x00007ffff789ca6f in __gnu_cxx::__verbose_terminate_handler () at /usr/src/debug/gcc/gcc/libstdc++-v3/libsupc++/vterminate.cc:95                                                                                                                                                                                         
#4  0x00007ffff78b011c in __cxxabiv1::__terminate (handler=<optimized out>) at /usr/src/debug/gcc/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48                                                                                                                                                                               
#5  0x00007ffff78b0189 in std::terminate () at /usr/src/debug/gcc/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:58                                                                                                                                                                                                               
#6  0x00007ffff78b03ed in __cxxabiv1::__cxa_throw (obj=<optimized out>, tinfo=0x7ffff7a6c0d0 <typeinfo for std::length_error>, dest=0x7ffff78c8580 <std::length_error::~length_error()>) at /usr/src/debug/gcc/gcc/libstdc++-v3/libsupc++/eh_throw.cc:98                                                                      
#7  0x00007ffff78a01ca in std::__throw_length_error (__s=__s@entry=0x7ffff79d5998 "basic_string::_M_create") at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/functexcept.cc:82                                                                                                                                               
#8  0x00007ffff79bce25 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create (this=0x7fffffff1ec0, __old_capacity=0, __capacity=<synthetic pointer>: 18446744073709551557) at /usr/src/debug/gcc/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.tcc:140         
#9  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char const*> (__end=0x5555f2d4c5f5 "", __beg=0x5555f2d4c630 "P\376\324\362UU", this=0x7fffffff1ec0) at /usr/src/debug/gcc/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.tcc:225                      
#10 std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string (__a=..., __n=<optimized out>, __s=<optimized out>, this=0x7fffffff1ec0) at /usr/src/debug/gcc/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.h:630
#11 std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string (this=0x7fffffff1ec0, __svw=..., __a=...) at /usr/src/debug/gcc/gcc-build/x86_64-pc-linux-gnu/libstdc++-v3/include/bits/basic_string.h:190
#12 0x000055555566daf3 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string<std::basic_string_view<char, std::char_traits<char> >, void> (this=0x7fffffff1ec0, __t=Python Exception <class 'OverflowError'>: int too big to convert
, __a=...) at /usr/include/c++/13.1.1/bits/basic_string.h:797
#13 0x0000555555b7fb8b in SURELOG::CompileHelper::adjustSize (this=0x55563fe2dc80, ts=0x5555fff49950, component=0x5555f2c11080, compileDesign=0x5555f10e8000, instance=0x5555fff2cf20, c=0x5555fc98cbe0, uniquify=true, sizeMode=false)
    at Surelog/src/DesignCompile/CompileHelper.cpp:3569
#14 0x0000555555c59053 in SURELOG::UhdmWriter::writeCont_assign (this=0x55563fe2dc70, netlist=0x5555fff3eea0, s=..., mod=0x5555f2c11080, m=0x55564261afc0, assigns=0x55564261dd00) at Surelog/src/DesignCompile/UhdmWriter.cpp:1765
#15 0x0000555555c636a1 in SURELOG::UhdmWriter::writeElabModule (this=0x55563fe2dc70, s=..., instance=0x5555fff2cf20, m=0x55564261afc0, exprBuilder=...) at Surelog/src/DesignCompile/UhdmWriter.cpp:3184
#16 0x0000555555c652b5 in SURELOG::UhdmWriter::writeInstance (this=0x55563fe2dc70, mod=0x5555f2c11080, instance=0x5555fff2cf20, m=0x55564261afc0, compileDesign=0x5555f10e8000, componentMap=std::map with 521 elements = {...}, modPortMap=std::map with 0 elements, instanceMap=std::map with 0 elements, exprBuilder=...)
    at Surelog/src/DesignCompile/UhdmWriter.cpp:3471
#17 0x0000555555c65dc4 in SURELOG::UhdmWriter::writeInstance (this=0x55563fe2dc70, mod=0x5555fff42580, instance=0x5555fff2cdc0, m=0x555642603800, compileDesign=0x5555f10e8000, componentMap=std::map with 521 elements = {...}, modPortMap=std::map with 0 elements, instanceMap=std::map with 0 elements, exprBuilder=...)
    at Surelog/src/DesignCompile/UhdmWriter.cpp:3553
#18 0x0000555555c661c1 in SURELOG::UhdmWriter::writeInstance (this=0x55563fe2dc70, mod=0x5555f2c39600, instance=0x5555fff2cc60, m=0x55564261ad80, compileDesign=0x5555f10e8000, componentMap=std::map with 521 elements = {...}, modPortMap=std::map with 0 elements, instanceMap=std::map with 0 elements, exprBuilder=...)
    at Surelog/src/DesignCompile/UhdmWriter.cpp:3599
#19 0x0000555555c65dc4 in SURELOG::UhdmWriter::writeInstance (this=0x55563fe2dc70, mod=0x5555fff28c00, instance=0x5555fff2c6e0, m=0x555642603680, compileDesign=0x5555f10e8000, componentMap=std::map with 521 elements = {...}, modPortMap=std::map with 0 elements, instanceMap=std::map with 0 elements, exprBuilder=...)
    at Surelog/src/DesignCompile/UhdmWriter.cpp:3553
#20 0x0000555555c661c1 in SURELOG::UhdmWriter::writeInstance (this=0x55563fe2dc70, mod=0x5555f2c39080, instance=0x5555fff2c580, m=0x55564261ab40, compileDesign=0x5555f10e8000, componentMap=std::map with 521 elements = {...}, modPortMap=std::map with 0 elements, instanceMap=std::map with 0 elements, exprBuilder=...)
    at Surelog/src/DesignCompile/UhdmWriter.cpp:3599
#21 0x0000555555c65dc4 in SURELOG::UhdmWriter::writeInstance (this=0x55563fe2dc70, mod=0x5555f2c5e580, instance=0x5555ffdb46e0, m=0x5556425ccb40, compileDesign=0x5555f10e8000, componentMap=std::map with 521 elements = {...}, modPortMap=std::map with 0 elements, instanceMap=std::map with 0 elements, exprBuilder=...)
    at Surelog/src/DesignCompile/UhdmWriter.cpp:3553
#22 0x0000555555c65dc4 in SURELOG::UhdmWriter::writeInstance (this=0x55563fe2dc70, mod=0x5555f2c84580, instance=0x5555fe9fc580, m=0x55564255e6c0, compileDesign=0x5555f10e8000, componentMap=std::map with 521 elements = {...}, modPortMap=std::map with 0 elements, instanceMap=std::map with 0 elements, exprBuilder=...)
    at Surelog/src/DesignCompile/UhdmWriter.cpp:3553
#23 0x0000555555c6a3cd in SURELOG::UhdmWriter::write (this=0x55563fe2dc70, uhdmFileId=...) at Surelog/src/DesignCompile/UhdmWriter.cpp:4133
#24 0x0000555555a4b8c3 in SURELOG::CompileDesign::writeUHDM (this=0x5555f10e8000, fileId=...) at Surelog/src/DesignCompile/CompileDesign.cpp:375
#25 0x00005555556e96d6 in SURELOG::Compiler::compile (this=0x5555570da000) at Surelog/src/SourceCompile/Compiler.cpp:1076
#26 0x0000555555671ad5 in SURELOG::start_compiler (clp=0x5555570d8000) at Surelog/src/API/Surelog.cpp:33
#27 0x000055555566affc in executeCompilation (argc=1041, argv=0x7fffffff4508, diffCompMode=false, fileUnit=false, overallStats=0x0) at Surelog/src/main.cpp:92
#28 0x000055555566c770 in main (argc=1041, argv=0x7fffffff4508) at Surelog/src/main.cpp:301
```
Last Surelog frame:
```
(gdb) f 13
#13 0x0000555555b7fb8b in SURELOG::CompileHelper::adjustSize (this=0x55563fe2dc80, ts=0x5555fff49950, component=0x5555f2c11080, compileDesign=0x5555f10e8000, instance=0x5555fff2cf20, c=0x5555fc98cbe0, uniquify=true, sizeMode=false)
[... cut ...] cpp:3569
3569                c->VpiValue("BIN:" + std::string(v));
```
Some debug information:
```
(gdb) p c->VpiValue()                                                                                                                                           
$12 = "BIN:0"
(gdb) p orig_size
$13 = 64
(gdb) p size
$14 = 4
```
The problematic line is: https://github.com/chipsalliance/Surelog/blob/master/src/DesignCompile/CompileHelper.cpp#LL3568C28-L3568C43, but I'm not sure how to properly fix it.

alaindargelas, 2023-05-08:
@kamilrakoczy please specify the full Opentitan context: repo and command line for Surelog.

alaindargelas, 2023-05-09:
Should be fixed by #3639, if not, please provide details above.

closed 2023-05-09 by alaindargelas

kamilrakoczy, 2023-05-09:
There is still segmentation fault while parsing opentitan after this, but this time it is in UHDM. I've prepared PR that fixes it: https://github.com/chipsalliance/UHDM/pull/932, but if this isn't proper fix (I'm not sure if there should be ``packed_array_var`` without any elements),

here are details how to reproduce it. Unfortunately process isn't straightforward, just opentitan repository isn't enough:

```
git clone https://github.com/antmicro/yosys-systemverilog
cd yosys-systemverilog/
git submodule update --init --recursive uhdm-tests/opentitan/opentitan
virtualenv venv-ot #optional step
source venv-ot/bin/activate #optional step
git apply ../opentitan_synth_patches/0015-Update-top-custom_tiny.patch ../opentitan_synth_patches/0016-Update-python-requirements.txt.patch
pip install -r python-requirements.txt
fusesoc --cores-root=. run --flag=fileset_top --flag=ot_is_custom_tiny --build --tool yosys --target=synth lowrisc:systems_custom_tiny:chip_custom_tiny_nexysvideo:0.1
cd build/lowrisc_systems_custom_tiny_chip_custom_tiny_nexysvideo_0.1/synth-yosys/
```
now, surelog command can be found in ``surelog.mk`` file, or alternative I'm posting it here (it is very long):

```
surelog -DSYNTHESIS -parse +define+AST_BYPASS_CLK=True+IBEX_CUSTOM_PMP_RESET_VALUES=True+PRIM_DEFAULT_IMPL=prim_pkg::ImplXilinx  -PBootRomInitFile=../../../../../../uhdm-tests/opentitan/opentitan_synth_vmem_files/test_rom_fpga_cw310.39.scr.vmem -POtpCtrlMemInitFile=../../../../../../uhdm-tests/opentitan/opentitan_synth_vmem_files/otp_img_fpga_cw310.vmem  -I../src/lowrisc_dv_dv_fcov_macros_0 -I../src/lowrisc_dv_dv_macros_0 -I../src/lowrisc_prim_assert_0.1/rtl -I../src/lowrisc_prim_util_get_scramble_params_0/rtl -I../src/lowrisc_prim_util_memload_0/rtl -I../src/lowrisc_ibex_ibex_core_0.1/rtl -I../src/lowrisc_ip_rv_core_ibex_0.1/rtl -sv ../src/lowrisc_constants_jtag_id_pkg_0/rtl/jtag_id_pkg.sv -sv ../src/lowrisc_constants_top_pkg_0/rtl/top_pkg.sv -sv ../src/lowrisc_dv_pins_if_0/pins_if.sv -sv ../src/lowrisc_ibex_ibex_pkg_0.1/rtl/ibex_pkg.sv -sv ../src/lowrisc_ip_entropy_src_pkg_0.1/rtl/entropy_src_pkg.sv -sv ../src/lowrisc_ip_i2c_pkg_0.1/rtl/i2c_pkg.sv -sv ../src/lowrisc_ip_jtag_pkg_0.1/rtl/jtag_pkg.sv -sv ../src/lowrisc_ip_spi_device_pkg_0.1/rtl/spi_device_reg_pkg.sv -sv ../src/lowrisc_ip_spi_device_pkg_0.1/rtl/spi_device_pkg.sv -sv ../src/lowrisc_ip_usbdev_pkg_0.1/rtl/usbdev_pkg.sv -sv ../src/lowrisc_prim_abstract_prim_pkg_0.1/prim_pkg.sv -sv ../src/lowrisc_prim_cipher_pkg_0.1/rtl/prim_cipher_pkg.sv -sv ../src/lowrisc_prim_crc32_0/rtl/prim_crc32.sv -sv ../src/lowrisc_prim_generic_and2_0/rtl/prim_generic_and2.sv -sv ../src/lowrisc_prim_generic_buf_0/rtl/prim_generic_buf.sv -sv ../src/lowrisc_prim_generic_clock_buf_0/rtl/prim_generic_clock_buf.sv -sv ../src/lowrisc_prim_generic_clock_gating_0/rtl/prim_generic_clock_gating.sv -sv ../src/lowrisc_prim_generic_flop_0/rtl/prim_generic_flop.sv -sv ../src/lowrisc_prim_generic_xnor2_0/rtl/prim_generic_xnor2.sv -sv ../src/lowrisc_prim_generic_xor2_0/rtl/prim_generic_xor2.sv -sv ../src/lowrisc_prim_otp_pkg_0.1/rtl/prim_otp_pkg.sv -sv ../src/lowrisc_prim_pad_wrapper_pkg_0/rtl/prim_pad_wrapper_pkg.sv -sv ../src/lowrisc_prim_ram_1p_pkg_0/rtl/prim_ram_1p_pkg.sv -sv ../src/lowrisc_prim_ram_2p_pkg_0/rtl/prim_ram_2p_pkg.sv -sv ../src/lowrisc_prim_rom_pkg_0/rtl/prim_rom_pkg.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_pkg.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_22_16_dec.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_22_16_enc.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_28_22_dec.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_28_22_enc.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_39_32_dec.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_39_32_enc.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_64_57_dec.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_64_57_enc.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_72_64_dec.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_72_64_enc.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_hamming_22_16_dec.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_hamming_22_16_enc.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_hamming_39_32_dec.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_hamming_39_32_enc.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_hamming_72_64_dec.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_hamming_72_64_enc.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_hamming_76_68_dec.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_hamming_76_68_enc.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_inv_22_16_dec.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_inv_22_16_enc.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_inv_28_22_dec.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_inv_28_22_enc.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_inv_39_32_dec.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_inv_39_32_enc.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_inv_64_57_dec.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_inv_64_57_enc.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_inv_72_64_dec.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_inv_72_64_enc.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_inv_hamming_22_16_dec.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_inv_hamming_22_16_enc.sv -sv ../src/lowrisc_prim_secded_0.1/rtl/prim_secded_inv_ha [... cut ...] /src/lowrisc_systems_custom_tiny_top_custom_tiny_0.1/rtl/autogen/top_custom_tiny_rnd_cnst_pkg.sv -sv ../src/lowrisc_systems_custom_tiny_top_custom_tiny_0.1/rtl/autogen/top_custom_tiny.sv -sv ../src/lowrisc_systems_custom_tiny_chip_custom_tiny_nexysvideo_0.1/rtl/clkgen_xil7series.sv -sv ../src/lowrisc_systems_custom_tiny_chip_custom_tiny_nexysvideo_0.1/rtl/usr_access_xil7series.sv -sv ../src/lowrisc_systems_custom_tiny_chip_custom_tiny_nexysvideo_0.1/rtl/autogen/chip_custom_tiny_nexysvideo.sv
```