chipsalliance/Surelog #3642: Missing ranges for struct typespec in union

kamilrakoczy, 2023-05-10:
Related to: https://github.com/chipsalliance/yosys-f4pga-plugins/issues/503
TEST:
```
package foo_flags;
  typedef struct packed {
    logic a;
    logic b;
    logic c;
  } common_flags_t;
endpackage : foo_flags

package fooes;

  typedef enum logic [1:0] {
    a,
    b,
    c,
    d
  } classes;

endpackage : fooes

package goog;

typedef struct packed {
  logic a;
  fooes::classes b;
} padded_fooes_t;

typedef union packed {
 foo_flags::common_flags_t  [3:0][7:0] atype_t;
 padded_fooes_t     [3:0][7:0] btype_t;
} top_flag_t;

endpackage: goog

module top ();
  goog::top_flag_t a;
endmodule
```
UHDM:
```
|uhdmtopPackages:
...
    |vpiTypedef:
    \_union_typespec: (goog::top_flag_t), line:27:9, endln:27:14
      |vpiParent:
      \_package: goog (goog::), file:dut.sv, line:20:1, endln:32:17
      |vpiName:goog::top_flag_t
      |vpiInstance:
      \_package: goog (goog::), file:dut.sv, line:20:1, endln:32:17
      |vpiPacked:1
      |vpiTypespecMember:
      \_typespec_member: (atype_t), line:28:40, endln:28:47
        |vpiParent:
        \_union_typespec: (goog::top_flag_t), line:27:9, endln:27:14
        |vpiName:atype_t
        |vpiTypespec:
        \_struct_typespec: (foo_flags::common_flags_t), line:2:11, endln:2:17
        |vpiRefFile:dut.sv
        |vpiRefLineNo:28
        |vpiRefColumnNo:2
        |vpiRefEndLineNo:28
        |vpiRefEndColumnNo:39
 ...
 ```
 There isn't information about ranges of ``atype_t`` (`` [3:0][7:0]``) member in UHDM.

alaindargelas, 2023-05-11:
Fixed by #3644

kamilrakoczy, 2023-05-15:
@alaindargelas I think ``atype_t`` ranges should be reported as ``vpiPackedArrayTypespec`` (packed ranges) instead of ``vpiArrayTypespec`` (unpacked ranges): https://github.com/chipsalliance/Surelog/pull/3644/files#diff-fd7b08ca9cc385e4e6986be492fa3ca51292171cdd9c1edfd44571f154f8b699R189.

Could you please confirm if this is the case?

alaindargelas, 2023-05-16:
Agreed, fixed by #3653

closed 2023-05-16 by alaindargelas