chipsalliance/Surelog #3655: Static size casting

grotival, 2023-05-16:
Reading the following verilog with UHDM then Yosys-f4pga results in the wrong values:

```
module static_size_casting (
    output wire [ 7:0] out1,
    output wire [ 7:0] out2,
    output wire [15:0] out3,
    output wire [15:0] out4,
    output wire [15:0] out5,
    output wire [15:0] out6
);

  // 8-bit wide outputs.
  assign out1 = 1 << 15;  // Expected 8'b00000000; OK
  assign out2 = 8'(1 << 15);  // Expected 8'b00000000; OK

  // 16-bit wide outputs.
  assign out3 = 1 << 15;  // Expected 16'b1000000000000000; OK
  assign out4 = 8'(1 << 15);  // Expected 16'b0000000000000000; NOT OK
  assign out5 = 16'(8'(1 << 15));  // Expected 16'b0000000000000000; NOT OK
  assign out6 = {8{2'(1'b1)}};  // Expected 16'b0101010101010101; NOT OK

endmodule : static_size_casting
```

Apologies if this bug is filed at the wrong place.

grotival, 2023-05-16:
FYI @hzeller @QuantamHD

alaindargelas, 2023-05-17:
@tgorochowik , 
To be noted, UHDM model is correct for all assigns.
It contains the complete operation expression tree.
It looks like either the plugin or Yosys needs help in reducing those to scalars.  
It is OK to do it in Surelog for now, as these end up being scalars in this testcase.
But if the same expression is a leaf in an expression that is overall not a constant, then
the same problem will arise, will the plugin/yosys be able to compute correctly those, example:

module static_size_casting (
    input wire [ 7:0] in7,
    output wire [15:0] out7
);
assign out7 = {{8{2'(1'b1)}}, in7}; 
endmodule

grotival, 2023-05-17:
Thanks!
Should I close this bug and open one on yosys-f4pga then or you're doing it or something else?

alaindargelas, 2023-05-17:
I'll fix the expression reduced as constants as in your original case, so you can leave this case open.
If you find a real live example as the one I proposed, then please file a new bug to the plugin first. We typically concentrate on real testcases from real designs. We don't try to manufacture and add support for synthetic tests. There is enough work as-is :-)

alaindargelas, 2023-05-18:
Fixed by #3662.
yosys-f4pga needs to update Surelog, @tgorochowik FYI.

closed 2023-05-18 by alaindargelas

grotival, 2023-05-22:
verified, all good now, thanks!

Just for documentation: this is still not working (but not worth fixing yet as it is not proven to be used IRL):
```systemverilog
module static_size_casting (
input wire [ 7:0] in7,
output wire [15:0] out7
);
assign out7 = {{4{2'(1'b1)}}, in7};
endmodule
```