chipsalliance/Surelog #3672: indices are mixed up when assigning values in array of array of packed struct

grotival, 2023-05-24:
Repro case:

```systemverilog
typedef struct packed {
  logic a;
  logic b;
} status_t;

module advanced (
    output status_t status_out00,
    output status_t status_out01,
    output status_t status_out10,
    output status_t status_out11
);

  // 2 x 2 x $bits(status_t) --> 8 bits
  status_t [1:0][1:0] group_status;

  assign group_status[0][0].a = 1'b1;
  assign group_status[0][0].b = 1'b0;
  // Expected 2'b10: OK
  assign status_out00 = group_status[0][0];

  assign group_status[0][1].a = 1'b1;
  assign group_status[0][1].b = 1'b0;
  // Expected 2'b10: OK
  assign status_out01 = group_status[0][1];

  assign group_status[1][0].a = 1'b1;
  assign group_status[1][0].b = 1'b0;
  // Expected 2'b10: NOT OK
  assign status_out10 = group_status[1][0];

  assign group_status[1][1].a = 1'b1;
  assign group_status[1][1].b = 1'b0;
  // Expected 2'b10: NOT OK
  assign status_out11 = group_status[1][1];

  // Expected group_status to be 8'b10101010 but got 8'bxxxx1010.

endmodule : advanced
```

Result is:
```systemverilog
  assign group_status = 8'bxxxx1010;
  assign status_out00 = 2'b10;
  assign status_out01 = 2'b10;
  assign status_out10 = 2'bxx;
  assign status_out11 = 2'bxx;
```

When looking more in details:
```
  assign group_status[1] = 1'b1;
  assign group_status[0] = 1'b0;
  assign status_out00 = group_status[1:0];
  assign group_status[1] = 1'b1;
  assign group_status[0] = 1'b0;
  assign status_out01 = group_status[3:2];
  assign group_status[3] = 1'b1;
  assign group_status[2] = 1'b0;
  assign status_out10 = group_status[5:4];
  assign group_status[3] = 1'b1;
  assign group_status[2] = 1'b0;
  assign status_out11 = group_status[7:6];
```

As usual, I'm doing:
SystemVerilog to UHDM using Surelog
UHDM to Yosys AST using Yosys f4pga
Yosys AST to Verilog

Apologies if the bug is on Yosys f4pga

grotival, 2023-05-24:
FYI @QuantamHD @hzeller

alaindargelas, 2023-05-25:
@tgorochowik , can you check too, the problem is likely in the plugin.

grotival, 2023-05-25:
thanks!

Adding a bit more details:

With verilog:
```systemverilog
typedef struct packed {
  logic a;
  logic b;
} status_t;

module array_of_array_of_packed_struct (
    output status_t status_out00,
    output status_t status_out01,
    output status_t status_out10,
    output status_t status_out11
);

  // 2 x 2 x $bits(status_t) --> 8 bits OK
  status_t [1:0][1:0] group_status;

  assign group_status[0][0].a = 1'b0;
  assign group_status[0][0].b = 1'b0;
  // Expected 2'b00: OK
  assign status_out00 = group_status[0][0];

  assign group_status[0][1].a = 1'b0;
  assign group_status[0][1].b = 1'b1;
  // Expected 2'b01: NOT OK
  assign status_out01 = group_status[0][1];

  assign group_status[1][0].a = 1'b1;
  assign group_status[1][0].b = 1'b0;
  // Expected 2'b10: NOT OK
  assign status_out10 = group_status[1][0];

  assign group_status[1][1].a = 1'b1;
  assign group_status[1][1].b = 1'b1;
  // Expected 2'b11: NOT OK
  assign status_out11 = group_status[1][1];

  // Expected group_status to be 8'b11100100 but got 8'bxxxx1000.

endmodule : array_of_array_of_packed_struct
```

I get:
```systemverilog

module array_of_array_of_packed_struct(status_out00, status_out01, status_out10, status_out11);
  wire [7:0] group_status;
  output [1:0] status_out00;
  wire [1:0] status_out00;
  output [1:0] status_out01;
  wire [1:0] status_out01;
  output [1:0] status_out10;
  wire [1:0] status_out10;
  output [1:0] status_out11;
  wire [1:0] status_out11;
  assign group_status = 8'bxxxx1000;
  assign status_out00 = 2'b00;
  assign status_out01 = 2'b10;
  assign status_out10 = 2'bxx;
  assign status_out11 = 2'bxx;
endmodule
```

or without flattening:

```systemverilog
/* Generated by 0.28+6 */

module array_of_array_of_packed_struct(status_out00, status_out01, status_out10, status_out11);
  wire [7:0] group_status;
  output [1:0] status_out00;
  wire [1:0] status_out00;
  output [1:0] status_out01;
  wire [1:0] status_out01;
  output [1:0] status_out10;
  wire [1:0] status_out10;
  output [1:0] status_out11;
  wire [1:0] status_out11;
  assign group_status[1] = 1'b0;
  assign group_status[0] = 1'b0;
  assign status_out00 = group_status[1:0];
  assign group_status[1] = 1'b0;
  assign group_status[0] = 1'b1;
  assign status_out01 = group_status[3:2];
  assign group_status[3] = 1'b1;
  assign group_status[2] = 1'b0;
  assign status_out10 = group_status[5:4];
  assign group_status[3] = 1'b1;
  assign group_status[2] = 1'b1;
  assign status_out11 = group_status[7:6];
endmodule
```

it seems like reading group_status with [x][y] works well but assigning doesn't. Likely a small silly bug?

alaindargelas, 2023-05-27:
@grotival, please re-open this case to the plugin Yosys f4pga.
The Surelog/UHDM tree is correct.
In general, you are better off (resolution time wise) opening your cases in the first place to the plugin since that is the output you observe.
If you were to observe the UHDM dump and see something incorrect there, then yes, please open the cases here, but since you observe the output of Synthesis, there are many steps that need to be checked bottom up (from the last step of the flow). 
Thanks

closed 2023-05-27 by alaindargelas