chipsalliance/Surelog #3708: Illegal lhs of type wire after: #3701

kamilrakoczy, 2023-06-14:
After https://github.com/chipsalliance/Surelog/pull/3701 some of previously valid assignments started to report error message:
```
Illegal lhs of type wire "wire name"
```
This occurs in both opentitan and ibex and can be reproduced using following example:
```
package ibex_pkg;                       
typedef struct packed {                 
  logic       irq_int;                              
  logic       irq_ext;                                 
  logic [4:0] lower_cause;
} exc_cause_t;                               
endpackage                                   
                                                                      
module dut(output ibex_pkg::exc_cause_t  exc_cause_o);
always_comb begin 
  exc_cause_o            = 6'h00; 
end 
endmodule
```
I think it is related to usage of type from package without import, as following example passes:
```
package ibex_pkg;
typedef struct packed {
  logic       irq_int;
  logic       irq_ext;
  logic [4:0] lower_cause;
} exc_cause_t;
endpackage

module dut import ibex_pkg::*;
(output exc_cause_t  exc_cause_o);

always_comb begin
  exc_cause_o            = 6'h00;
end
endmodule
```

It also can be seen in PR logs: https://github.com/chipsalliance/Surelog/pull/3701/commits/ecd75256486e2a50fd73d690a2fe54f1fc6fa821#diff-5a24425284e58c24f675c3bd9ba27255045d189229ffb0d42fd4a3f620196b69R39348

Thomasb81, 2023-06-14:
It looks like 
```
          if (tmp->UhdmType() == uhdmalways || tmp->UhdmType() == uhdminitial) {
```
has missing case : `always_comb`, `always_latch` and `always_ff` may be even `final` ?

Thomasb81, 2023-06-15:
Right. 
So need to understand why `tmp->UhdmType() == udhmalways` expression is not true in this case.

> `final` is modeled as `final_stmt`. Please confirm if this should be included in the implemented test.

I don't get your point. I can't find in the standard `final_stmt`. The standard specify `#define vpiFinal 676` 

Chapter 9.2.3 discuss final. It says it is like initial. But is call at end of simulation time. So usage is limited to do to print and/or thing that do not consume time...
So from a linter perspective if we identify that parent of an assignment statement is a final process, we should probably report an issue.

Thomasb81, 2023-06-15:
> This use case you presented here, the assignment to _exc_cause_o_ is in the `always` block and it's a wire. And, so the reported error.

With a reference  simulator the code does not report any issue to elaborate the testcase. So I think the source code is legal. Also I think a bunch of other commercial tool has already process this code ...

Additionnally we can observe with a "reference simulator"
Is legal
```
module dut (output logic o);
always_comb begin
    o = 1'b0;
end
endmodule
```
Is not legal
```
module dut (output wire logic o);
always_comb begin
    o = 1'b0;
end
endmodule
```
Same with orignal testcase modified to add the wire : it become illegal
```
module dut(output wire ibex_pkg::exc_cause_t  exc_cause_o);
always_comb begin 
  exc_cause_o            = 6'h00; 
end 
endmodule
```
I guess `output logic o` is a shorthand for `output var logic o`...
I should admit as an user I never take care to fully understand the difference between a net and a variable.

I seems to me to have understand from Surlog source code on port creation that it is currently systematically creating both object net and variable, is that right ? But probably on this case the port creation should not imply a net but only a variable...

So in other words, the lint check is probably ok if port is associate to a wire because the assignment cannot be in a procedural assignment. But a port may not be a net, and in this case the error must not fire.

Also to complete the lint check other kind of standard nettype need to be considered (tri, tri0,supply0,wand,triand,tri1,supply1,wor,trior,trireg,uwire and user-defined nettype) but that's more a concern for simulation than synthesis use-case.

There is probably something else if tool behavior also change with or without import as mention by @kamilrakoczy

alaindargelas, 2023-06-17:
Fixed by #3716

closed 2023-06-17 by alaindargelas