chipsalliance/Surelog #3713: Regression: wrong const evaluation of unsized const

kamilrakoczy, 2023-06-15:
After #3701 Surelog started to report unsized assign as ``1'b1`` in ``(out3 == '1``.

TEST:
```
typedef struct packed {logic [3:0] a;} my_struct_packed_t;

module top (
    output wire [3:0] out1,
    output wire out2,
    output my_struct_packed_t out3,
    output wire out4
);
  assign out1   = '1;
  assign out2   = (out1 == 4'b1111);
  assign out3.a = '1;
  assign out4   = (out3 == '1);
endmodule : top
```

UHDM (using current master: af8144d5a9643209d7ba2cd0bc6331b266d72c91):
```
...
  |vpiContAssign:
  \_cont_assign: , line:12:10, endln:12:31
    |vpiParent:
    \_module_inst: work@top (work@top), file:UHDM-integration-tests/tests/UnsizedConst/top.sv, line:3:1, endln:13:16
    |vpiRhs:
    \_operation: , line:12:20, endln:12:30
      |vpiParent:
      \_cont_assign: , line:12:10, endln:12:31
      |vpiOpType:14
      |vpiOperand:
      \_ref_obj: (work@top.out3), line:12:20, endln:12:24
        |vpiParent:
        \_operation: , line:12:20, endln:12:30
        |vpiName:out3
        |vpiFullName:work@top.out3
        |vpiActual:
        \_logic_net: (work@top.out3), line:6:31, endln:6:35
      |vpiOperand:
      \_constant: , line:12:28, endln:12:30
        |vpiDecompile:1
        |vpiSize:1
        |UINT:1
        |vpiConstType:9
```
Previously Surelog left it as unsized in UHDM:
```
  |vpiContAssign:
  \_cont_assign: , line:12:10, endln:12:31
    |vpiRhs:
    \_operation: , line:12:20, endln:12:30
      |vpiParent:
      \_cont_assign: , line:12:10, endln:12:31
      |vpiOpType:14
      |vpiOperand:
      \_ref_obj: (out3), line:12:20, endln:12:24
        |vpiParent:
        \_operation: , line:12:20, endln:12:30
        |vpiName:out3
      |vpiOperand:
      \_constant: , line:12:28, endln:12:30
        |vpiParent:
        \_operation: , line:12:20, endln:12:30
        |vpiDecompile:'1
        |vpiSize:-1
        |BIN:1
        |vpiConstType:3
  ```

kamilrakoczy, 2023-06-16:
I think this is still not fixed.
I've tested it using current Surelog master (https://github.com/chipsalliance/Surelog/commit/72d22b7d918962830eac88a5473be8440633f9d2).
With this version, I still have comparison of ``1'b1`` to ``out3``:
```
  |vpiContAssign:                                                                                                                            
  \_cont_assign: , line:12:10, endln:12:31
    |vpiParent:                   
    \_module_inst: work@top (work@top), file:/home/grey/uhdm/yosys-uhdm-plugin-integration/UHDM-integration-tests/tests/t/dut.sv, line:3:1, endln:13:16
    |vpiRhs:                         
    \_operation: , line:12:20, endln:12:30                                                                                                   
      |vpiParent:                                                                                                                                                                                                                                                                         
      \_cont_assign: , line:12:10, endln:12:31                                                                                               
      |vpiOpType:14                                                                                                                                                                                                                                                                       
      |vpiOperand:                                                                                                                                                                                                                                                                        
      \_ref_obj: (work@top.out3), line:12:20, endln:12:24                                                                                                                                                                                                                                 
        |vpiParent:                                                                                                                                                                                                                                                                       
        \_operation: , line:12:20, endln:12:30
        |vpiName:out3                                                                                                                                                                                                                                                                     
        |vpiFullName:work@top.out3                                                                                                           
        |vpiActual:                                                                                                                          
        \_logic_net: (work@top.out3), line:6:31, endln:6:35                                                                                                                                                                                                                               
      |vpiOperand:                                                                                                                                                                                                                                                                        
      \_constant: , line:12:28, endln:12:30                                                                                                                                                                                                                                               
        |vpiDecompile:1                                                                                                                      
        |vpiSize:1                                                                                                                                                                                                                                                                        
        |UINT:1                                                                                                                                                                                                                                                                           
        |vpiConstType:9                                                                                                                      
    |vpiLhs:                                                                                                                                                                                                                                                                              
    \_ref_obj: (work@top.out4), line:12:10, endln:12:14                                                                                      
      |vpiParent:                                                                                                                                                                                                                                                                         
      \_cont_assign: , line:12:10, endln:12:31
      |vpiName:out4
      |vpiFullName:work@top.out4
      |vpiActual:
      \_logic_net: (work@top.out4), line:7:17, endln:7:21
```
Note: the problem is with:
```
assign out4   = (out3 == '1);
```
Surelog correctly evaluates value for:
```
assign out3.a = '1;
```
or
```
  assign out1   = '1;
  assign out2   = (out1 == 4'b1111);
```

kamilrakoczy, 2023-06-16:
Oh, I think I know what caused confusion. I'm interested in elaborated tree (``uhdmtopModules``). I can see that in unelaborated tree (``uhdmallModules``) there is indeed unsized const: https://github.com/chipsalliance/Surelog/blob/master/tests/HierPathCont/HierPathCont.log#L485

alaindargelas, 2023-06-18:
Fixed by #3719.
Please note that this unsized const expansion is only safe to do when the -elabuhdm option is used.
If it is not used, the Adjustement is no longer invoked and the const remains unsized.

closed 2023-06-18 by alaindargelas