chipsalliance/Surelog #3724: Unresolved hierarchical reference when using -elabuhdm

kamilrakoczy, 2023-06-19:
TEST:
```
package pkg;
   typedef struct packed {
      logic       x;
   } a;
   typedef a[5:0] b;
endpackage
   
module top(output o);
   pkg::b c;
   assign c[1].x = 1;
   assign o = c[1].x;   
endmodule
```
Command:
```
surelog -parse -sverilog -elabuhdm top.sv
```
Error:
```
[ERR:UH0725] top.sv:10:11: Unresolved hierarchical reference "c[1].x".

[ERR:UH0725] top.sv:11:15: Unresolved hierarchical reference "c[1].x".
```

kamilrakoczy, 2023-06-19:
More tests with the same error (if you prefer, I can create separate issues for them):
```
interface sim_sram_if;   
   int start_addr;
endinterface // sim_sram_if

module dut(sim_sram_if ss_if);
   assign ss_if.start_addr = 32'h1234;
endmodule // dut

module top(output int o);
   sim_sram_if u_sim_sram_if();

   dut u_dut(.ss_if(u_sim_sram_if));

   assign o = u_sim_sram_if.start_addr;
endmodule // top
```
```
interface sim_sram_if;   
   int start_addr;
endinterface // sim_sram_if

module dut(sim_sram_if ss_if, output int a);
   assign a = ss_if.start_addr;
endmodule // dut

module top(output int o);
   sim_sram_if u_sim_sram_if();
   assign u_sim_sram_if.start_addr = 32'h1234;
   dut u_dut(.ss_if(u_sim_sram_if), .a(o));
endmodule // top
```

kamilrakoczy, 2023-06-19:
```systemverilog
interface Foo;
    parameter MULTIPLIER = 7;
    function integer bar;
        input integer x;
        return x * x * MULTIPLIER;
    endfunction
endinterface

module top;
    Foo foo_1();
    Foo #(.MULTIPLIER(8)) foo_2();
    initial begin
        $display(foo_1.bar(3));
        $display(foo_2.bar(3));
    end
endmodule
```
```systemverilog
`default_nettype none
module genblk_dive_top(output wire x);
        generate
                if (1) begin : Z
                        if (1) begin : A
                                wire x;
                                if (1) begin : B
                                        wire x;
                                        if (1) begin : C
                                                wire x;
                                                assign B.x = 0;
                                                wire z = A.B.C.x;
                                        end
                                        assign A.x = A.B.C.x;
                                end
                                assign B.C.x = B.x;
                        end
                end
        endgenerate
        assign x = Z.A.x;
endmodule
```
```systemverilog
module carryadd(a, b, y);

parameter WIDTH = 8;

input [WIDTH-1:0] a, b;
output [WIDTH-1:0] y;

genvar i;
generate
        for (i = 0; i < WIDTH; i = i+1) begin:STAGE
                wire IN1 = a[i], IN2 = b[i];
                wire C, Y;
                if (i == 0)
                        assign C = IN1 & IN2, Y = IN1 ^ IN2;
                else
                        assign C = (IN1 & IN2) | ((IN1 | IN2) & STAGE[i-1].C),
                               Y = IN1 ^ IN2 ^ STAGE[i-1].C;
                assign y[i] = Y;
        end
endgenerate

// assert property (y == a + b);

endmodule
```
```systemverilog
`default_nettype none
module named_genblk_top;
        generate
                if (1) begin
                        wire t;
                        begin : foo
                                wire x;
                        end
                        wire u;
                end
                begin : bar
                        wire x;
                        wire y;
                        begin : baz
                                wire x;
                                wire z;
                        end
                end
        endgenerate
        assign genblk1.t = 1;
        assign genblk1.foo.x = 1;
        assign genblk1.u = 1;
        assign bar.x = 1;
        assign bar.y = 1;
        assign bar.baz.x = 1;
        assign bar.baz.z = 1;
endmodule
```
```systemverilog
module loop_var_shadow_top(out);
        genvar i;
        generate
                for (i = 0; i < 2; i = i + 1) begin : loop
                        localparam j = i + 1;
                        if (1) begin : blk
                                localparam i = j + 1;
                                wire [i:0] x;
                                assign x = 1'sb1;
                        end
                end
        endgenerate
        output wire [63:0] out;
        assign out = {loop[0].blk.x, loop[1].blk.x};
endmodule
```

kamilrakoczy, 2023-06-19:
I believe the priority is quite high, as it prevents us from updating Surelog in our flow (and not updating Surelog block us with another fixes as they require newer Surelog).

alaindargelas, 2023-06-19:
@hs-apotell this is not due to late binding.
Some of your recent changes about parenting did break the hier_path::DeepClone method.

I'm fixing it.

alaindargelas, 2023-06-19:
#3725 fixes the first testcase in this Issue. Similar fixes are still needed for all other cases.

alaindargelas, 2023-06-20:
#3726 fixes 2 interface tests

alaindargelas, 2023-06-20:
#3727 fixes the last interface example

alaindargelas, 2023-06-21:
#3729 fixes genblk_dive_top

alaindargelas, 2023-06-21:
#3730 fixes carryadd

alaindargelas, 2023-06-27:
Putting a pause on  named_genblk_top which requires elaboration/grammar changes.

mglb, 2023-07-06:
There are also 2 tests from Surelog itself that fail with this error:
`surelog -parse -sverilog -elabuhdm Surelog/tests/OneNetRange/dut.v`
`surelog -parse -sverilog -elabuhdm Surelog/tests/OneNetInterf/dut.v`

alaindargelas, 2023-07-07:
Fixed named_genblk_top with https://github.com/chipsalliance/Surelog/pull/3744

alaindargelas, 2023-07-07:
@mglb these last 2 tests never worked, you can ignore.
I'll work on them later.

alaindargelas, 2023-07-10:
#3749 fixes the last 2 cases

alaindargelas, 2023-07-21:
core/empty_args_hier.v Fixed by #3766

alaindargelas, 2023-07-22:
TypedefVariableDimensions/top.sv Fixed by #3767

alaindargelas, 2023-07-22:
core/empty_args_hier.sv Fixed by #3769

closed 2023-07-22 by alaindargelas