chipsalliance/Surelog #3776: surelog renaming wires with escape characters with #~@

gadfort, 2023-07-26:
Surelog is renaming some wires with `#~@` at the start and end of the name. However this is only in the declaration of the wire and not anywhere else (from what I can tell).

Version: VERSION: 1.65 BUILT  : Jun 20 2023

Original file:
```
line 517:
  wire [11:0] \mc_proc.ts_mc_proc_local_2_bits ;
```
Surelog output:
```
line 517
  wire [11:0] #~@mc_proc.ts_mc_proc_local_2_bits#~@;
```

To run: [encode_stream_sc_module_8.zip](https://github.com/chipsalliance/Surelog/files/12172964/encode_stream_sc_module_8.zip)
```
surelog -parse encode_stream_sc_module_8.v
```
output file: slpp_all/lib/work/encode_stream_sc_module_8.v with modified declarations

alaindargelas, 2023-07-26:
Correct, that is a feature.
The preprocessor output that you are looking at contains that transformation.
The Surelog parser does the inverse transform to parse the design, it you look at the UHDM output, you'll see the original names.
What are you trying to do with the output of the preprocessor?
It looks like you are parsing (-parse) and not only preprocessing, so this transform is necessary.

gadfort, 2023-07-26:
@alaindargelas we were using it to build the output file for other tools (https://github.com/chipsalliance/Surelog/issues/1908).
Using surelog to generate the list of files needed with preprocessing handled.
then concatenating these files together into a single file.
We are not using UHDM since it's not well supported, but it looks like for what we want, we cannot use these files safely and I can't see an option for surelog to output a regular verilog file at the end. Is there an option for that? or is UHDM the only choice?

alaindargelas, 2023-07-26:
Nothing has changed there for over 5 years, Surelog always dealt with escaped names with those symbols.
You noticing the issue only now, means you didn't have testcases with escaped names until now.
Surelog needs to have those symbols in the post-preprocessor output so it can parse the design, elaborate and extract the list of files you need to perform this task https://github.com/chipsalliance/Surelog/issues/1908, so those symbols are not going away.
A simple post-processing of the files in the filelist (sed) on the siliconcompiler side will solve the problem:

Replacing #~@([a-zA-Z_0-9\.]*)#~@   by   \<name> (space)

gadfort, 2023-07-26:
@alaindargelas you are correct, it would seem that we managed to avoid this issue until now.
Are there any other patterns we should be aware of?

gadfort, 2023-07-26:
@nmoroze FYI

alaindargelas, 2023-07-27:
@gadfort, correct, that is the only transform you have to worry about.

gadfort, 2023-07-27:
@alaindargelas thank you for your help and guidance

closed 2023-07-27 by gadfort