chipsalliance/Surelog #3789: Missing declarations of genvar in unnamed generate-for loops

mglb, 2023-08-04:
Test case: https://github.com/YosysHQ/yosys/blob/f37ce5c839c5a5e21fb74046375e5928ad91d22e/tests/simple/constpower.v

command: `surelog -elabuhdm -parse -d uhdm yosys/tests/simple/constpower.v`

UHDM contains references to `i` where they are expected, but there are no declarations.
The `j` used in the inner named loop has both references and parameter declarations.
Everything works as expected when the outer loop is changed to have a name.

<details><summary>Fragment of an UHDM dump (first vpiGenScopeArray in uhdmtopModules)</summary>

```
  |vpiGenScopeArray:
  \_gen_scope_array: (work@constpower.V[0]), line:9:30, endln:12:5
    |vpiParent:
    \_module_inst: work@constpower (work@constpower), file:/constpower.v, line:1:1, endln:15:10
    |vpiName:V[0]
    |vpiFullName:work@constpower.V[0]
    |vpiGenScope:
    \_gen_scope: (work@constpower.V[0]), line:9:30, endln:12:5
      |vpiParent:
      \_gen_scope_array: (work@constpower.V[0]), line:9:30, endln:12:5
      |vpiFullName:work@constpower.V[0]
      |vpiParameter:
      \_parameter: (work@constpower.V[0].j), line:9:0
        |vpiParent:
        \_gen_scope: (work@constpower.V[0]), line:9:30, endln:12:5
        |UINT:0
        |vpiTypespec:
        \_int_typespec:
        |vpiLocalParam:1
        |vpiName:j
        |vpiFullName:work@constpower.V[0].j
      |vpiContAssign:
      \_cont_assign: , line:10:10, endln:10:72
        |vpiParent:
        \_gen_scope: (work@constpower.V[0]), line:9:30, endln:12:5
        |vpiRhs:
        \_constant:
          |vpiParent:
          \_cont_assign: , line:10:10, endln:10:72
          |vpiDecompile:0
          |vpiSize:64
          |INT:0
          |vpiConstType:7
        |vpiLhs:
        \_part_select: ys (work@constpower.ys), line:10:10, endln:10:41
          |vpiParent:
          \_cont_assign: , line:10:10, endln:10:72
          |vpiName:ys
          |vpiFullName:work@constpower.ys
          |vpiDefName:ys
          |vpiActual:
          \_logic_net: (work@constpower.ys), line:1:19, endln:1:21
          |vpiConstantSelect:1
          |vpiLeftRange:
          \_operation: , line:10:13, endln:10:27
            |vpiParent:
            \_part_select: ys (work@constpower.ys), line:10:10, endln:10:41
            |vpiOpType:24
            |vpiOperand:
            \_operation: , line:10:13, endln:10:23
              |vpiParent:
              \_operation: , line:10:13, endln:10:27
              |vpiOpType:24
              |vpiOperand:
              \_operation: , line:10:13, endln:10:16
                |vpiParent:
                \_operation: , line:10:13, endln:10:23
                |vpiOpType:25
                |vpiOperand:
[... 190 lines cut ...]
                |vpiConstType:9
            |vpiOperand:
            \_operation: , line:11:36, endln:11:40
              |vpiParent:
              \_operation: , line:11:30, endln:11:40
              |vpiOpType:25
              |vpiOperand:
              \_ref_obj: (work@constpower.V[0].yu.j), line:11:36, endln:11:37
                |vpiParent:
                \_operation: , line:11:36, endln:11:40
                |vpiName:j
                |vpiFullName:work@constpower.V[0].yu.j
                |vpiActual:
                \_parameter: (work@constpower.V[0].j), line:9:0
              |vpiOperand:
              \_constant: , line:11:38, endln:11:40
                |vpiParent:
                \_operation: , line:11:36, endln:11:40
                |vpiDecompile:64
                |vpiSize:64
                |UINT:64
                |vpiConstType:9
```

</details>

alaindargelas, 2023-08-05:
Note that Surelog never reported the parameter for i, so this is not a regression.

Also note that if you change the testcase to have explicit blocks, both loop indexes get reported as parameters:

generate
	for (i = 0; i < 2; i = i+1) begin:V
	for (j = 0; j < 2; j = j+1) begin:W
		assign ys= 1;
		
	end
end
endgenerate

alaindargelas, 2023-08-07:
Fixed by #3791

closed 2023-08-07 by alaindargelas