chipsalliance/Surelog #3790: How does Surelog handle duplicate module names between sources and libraries?

nmoroze, 2023-08-04:
Let's say I have a setup like so:

```shell
$  tree 
.
├── lib
│   └── foo.v
└── foo.v
```

where the `foo.v` files contain conflicting module names `foo`. If I invoke Surelog like so:

```shell
$ surelog -parse -y lib foo.v -top foo
```

I notice that Surelog points to the top-level module as being in the file explicitly provided as a source:

```
[NTE:EL0503] /tmp/test/foo.v:1:1: Top level module "work@foo".
```

However, if I look at the elaborated file list, it includes the one in the library instead:

```shell
$ cat slpp_all/file_elab.lst
/tmp/test/slpp_all/lib/work/foo.v
```

I'm curious what sort of guarantees, if any, Surelog is meant to provide in this sort of situation regarding which file it ends up elaborating?

alaindargelas, 2023-08-05:
No guarantees, the standard actually says the last file parsed wins, but in the case of source files and library files I don't even know what is expected.

nmoroze, 2023-08-07:
Thank you for the clarification!

closed 2023-08-07 by nmoroze