chipsalliance/Surelog #3794: Incorrect names are assigned to unnamed generate-for scopes in some cases

mglb, 2023-08-07:
Used command:
```
surelog -parse -elabuhdm -d uhdm dut.v | sed -n -e '/Compile generate block/ p' -e '/uhdmtopModules/,$ { /\\_gen_scope\|BIN:/ p }'
```
(filtering for `BIN:` to show where `assing out[...]` happens)

Note that log messages in all cases refer to correct/expected hierarchical names.

The "for for begin" case is the most important one, as this is a situation from the test shown in #3789, which is failing in the plugin's CI.
Yosys only keeps last scope from a set of scopes with the same name. The rest, and `assign`s inside them, are dropped.
I could make a work-around in the plugin and fix repeated scope naming there, but I believe it should be fixed in Surelog.

## "for begin for begin": OK

Input:
```systemverilog
module dut(out);
output logic [2*2*2-1:0] out;
genvar i, j;
generate
    for (i = 0; i < 2; i = i+1) begin
    for (j = 0; j < 2; j = j+1) begin
        assign out[i*2 + j*4 + 1 : i*2 + j*4] = {i[0], j[0]};
    end
    end
endgenerate
endmodule
```

Result:
```
[INF:CP0335] dut.v:5:30: Compile generate block "work@dut.genblk1[0]".
[INF:CP0335] dut.v:6:30: Compile generate block "work@dut.genblk1[0].genblk1[0]".
[INF:CP0335] dut.v:6:30: Compile generate block "work@dut.genblk1[0].genblk1[1]".
[INF:CP0335] dut.v:5:30: Compile generate block "work@dut.genblk1[1]".
[INF:CP0335] dut.v:6:30: Compile generate block "work@dut.genblk1[1].genblk1[0]".
[INF:CP0335] dut.v:6:30: Compile generate block "work@dut.genblk1[1].genblk1[1]".
  \_gen_scope_array: (work@dut.genblk1[0]), line:5:30, endln:9:5
    \_gen_scope: (work@dut.genblk1[0]), line:5:30, endln:9:5
      \_gen_scope_array: (work@dut.genblk1[0]), line:5:30, endln:9:5
        \_gen_scope: (work@dut.genblk1[0]), line:5:30, endln:9:5
      \_gen_scope_array: (work@dut.genblk1[0].genblk1[0]), line:6:30, endln:8:5
        \_gen_scope: (work@dut.genblk1[0]), line:5:30, endln:9:5
        \_gen_scope: (work@dut.genblk1[0].genblk1[0]), line:6:30, endln:8:5
          \_gen_scope_array: (work@dut.genblk1[0].genblk1[0]), line:6:30, endln:8:5
            \_gen_scope: (work@dut.genblk1[0].genblk1[0]), line:6:30, endln:8:5
            \_gen_scope: (work@dut.genblk1[0].genblk1[0]), line:6:30, endln:8:5
              |BIN:00
      \_gen_scope_array: (work@dut.genblk1[0].genblk1[1]), line:6:30, endln:8:5
        \_gen_scope: (work@dut.genblk1[0]), line:5:30, endln:9:5
        \_gen_scope: (work@dut.genblk1[0].genblk1[1]), line:6:30, endln:8:5
          \_gen_scope_array: (work@dut.genblk1[0].genblk1[1]), line:6:30, endln:8:5
            \_gen_scope: (work@dut.genblk1[0].genblk1[1]), line:6:30, endln:8:5
            \_gen_scope: (work@dut.genblk1[0].genblk1[1]), line:6:30, endln:8:5
              |BIN:01
  \_gen_scope_array: (work@dut.genblk1[1]), line:5:30, endln:9:5
    \_gen_scope: (work@dut.genblk1[1]), line:5:30, endln:9:5
      \_gen_scope_array: (work@dut.genblk1[1]), line:5:30, endln:9:5
        \_gen_scope: (work@dut.genblk1[1]), line:5:30, endln:9:5
      \_gen_scope_array: (work@dut.genblk1[1].genblk1[0]), line:6:30, endln:8:5
        \_gen_scope: (work@dut.genblk1[1]), line:5:30, endln:9:5
        \_gen_scope: (work@dut.genblk1[1].genblk1[0]), line:6:30, endln:8:5
          \_gen_scope_array: (work@dut.genblk1[1].genblk1[0]), line:6:30, endln:8:5
            \_gen_scope: (work@dut.genblk1[1].genblk1[0]), line:6:30, endln:8:5
            \_gen_scope: (work@dut.genblk1[1].genblk1[0]), line:6:30, endln:8:5
              |BIN:10
      \_gen_scope_array: (work@dut.genblk1[1].genblk1[1]), line:6:30, endln:8:5
        \_gen_scope: (work@dut.genblk1[1]), line:5:30, endln:9:5
        \_gen_scope: (work@dut.genblk1[1].genblk1[1]), line:6:30, endln:8:5
          \_gen_scope_array: (work@dut.genblk1[1].genblk1[1]), line:6:30, endln:8:5
[... 185 lines cut ...]
      \_gen_scope_array: (work@dut.genblk1[0]), line:7:3, endln:7:56
        \_gen_scope: (work@dut.genblk1[0]), line:7:3, endln:7:56
        \_gen_scope: (work@dut.genblk1[0]), line:7:3, endln:7:56
        \_gen_scope: (work@dut.genblk1[0]), line:7:3, endln:7:56
          |BIN:00
  \_gen_scope_array: (work@dut.genblk1[0].genblk1[1]), line:7:3, endln:7:56
    \_gen_scope: (work@dut.genblk1[1]), line:7:3, endln:7:56
      \_gen_scope_array: (work@dut.genblk1[0].genblk1[1]), line:7:3, endln:7:56
        \_gen_scope: (work@dut.genblk1[1]), line:7:3, endln:7:56
        \_gen_scope: (work@dut.genblk1[1]), line:7:3, endln:7:56
          |BIN:01
  \_gen_scope_array: (work@dut.genblk1[1].genblk1[0]), line:7:3, endln:7:56
    \_gen_scope: (work@dut.genblk1[0]), line:7:3, endln:7:56
      \_gen_scope_array: (work@dut.genblk1[1].genblk1[0]), line:7:3, endln:7:56
        \_gen_scope: (work@dut.genblk1[0]), line:7:3, endln:7:56
        \_gen_scope: (work@dut.genblk1[0]), line:7:3, endln:7:56
          |BIN:10
  \_gen_scope_array: (work@dut.genblk1[1]), line:7:3, endln:7:56
    \_gen_scope: (work@dut.genblk1[1]), line:7:3, endln:7:56
      \_gen_scope_array: (work@dut.genblk1[1]), line:7:3, endln:7:56
        \_gen_scope: (work@dut.genblk1[1]), line:7:3, endln:7:56
        \_gen_scope: (work@dut.genblk1[1]), line:7:3, endln:7:56
        \_gen_scope: (work@dut.genblk1[1]), line:7:3, endln:7:56
          |BIN:11
```

alaindargelas, 2023-08-08:
@mglb @tgorochowik , please do not block the upgrade of the plugin because of this issue.
We need to get the plugin upgraded with Surelog and pass the https://github.com/chipsalliance/Surelog/actions/runs/5790010292 test ASAP.
I'll work on the issues above in the next few weeks, the regression is due to the new grammar used for the implicit generate blocks. 
It is not that important, that kind of code is not in any of the testcases we care about.

alaindargelas, 2023-08-08:
Actually I think there was a simpler fix than what I thought.
Fixed by #3797.
Let me know.

closed 2023-08-08 by alaindargelas

mglb, 2023-08-29:
Thanks, the test this came from started to pass.