chipsalliance/Surelog #3803: Parameter initialization with array assignment pattern containing `default` produces invalid elaborated UHDM

mglb, 2023-08-22:
## Problematic test case

```systemverilog
module test #(
  parameter int BobaCount = 2,
  parameter int unsigned NumBobaDrinkers[BobaCount] = '{default: '1}
)();
endmodule
```

UHDM:
```
|uhdmtopModules:
\_module_inst: work@test (work@test), file:/home/user/p/repos/yosys-systemverilog/test1.sv, line:1:1, endln:5:10
  ...
  |vpiParamAssign:
  \_param_assign: , line:3:26, endln:3:69
    |vpiParent:
    \_module_inst: work@test (work@test), file:/home/user/p/repos/yosys-systemverilog/test1.sv, line:1:1, endln:5:10
    |vpiRhs:
    \_array_expr:
      |vpiParent:
      \_param_assign: , line:3:26, endln:3:69
      |vpiTypespec:
      \_array_typespec: , line:3:13, endln:3:52
      |vpiExpr:
      \_constant:
        |vpiDecompile:1
        |UINT:1
        |vpiConstType:9
      |vpiExpr:
      \_constant:
    |vpiLhs:
    \_parameter: (work@test.NumBobaDrinkers), line:3:26, endln:3:41
```
Issue: last constant and array_typespec are empty

## Similar test case that works (1)

```systemverilog
module test #(
  parameter int BobaCount = 2,
  parameter int unsigned NumBobaDrinkers[BobaCount] = '{'1, '1}
)();
endmodule
```

UHDM:
```
|uhdmtopModules:
\_module_inst: work@test (work@test), file:/home/user/p/repos/yosys-systemverilog/test2.sv, line:1:1, endln:5:10
  ...
  |vpiParamAssign:
  \_param_assign: , line:3:26, endln:3:64
    |vpiParent:
    \_module_inst: work@test (work@test), file:/home/user/p/repos/yosys-systemverilog/test2.sv, line:1:1, endln:5:10
    |vpiRhs:
    \_operation: , line:3:55, endln:3:64
      |vpiParent:
      \_param_assign: , line:3:26, endln:3:64
      |vpiTypespec:
      \_array_typespec: , line:3:13, endln:3:52
        |vpiParent:
        \_operation: , line:3:55, endln:3:64
        |vpiRange:
        \_range: , line:3:42, endln:3:51
          |vpiParent:
          \_array_typespec: , line:3:13, endln:3:52
          |vpiLeftRange:
          \_constant:
            |vpiParent:
            \_range: , line:3:42, endln:3:51
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
            |vpiConstType:9
          |vpiRightRange:
          \_operation: , line:3:42, endln:3:51
            |vpiParent:
            \_range: , line:3:42, endln:3:51
            |vpiOpType:11
            |vpiOperand:
            \_ref_obj: (work@test.BobaCount), line:3:42, endln:3:51
              |vpiParent:
              \_operation: , line:3:42, endln:3:51
              |vpiName:BobaCount
              |vpiFullName:work@test.BobaCount
              |vpiActual:
              \_parameter: (work@test.BobaCount), line:2:17, endln:2:26
            |vpiOperand:
            \_constant:
              |vpiParent:
              \_operation: , line:3:42, endln:3:51
              |vpiSize:64
              |INT:1
              |vpiConstType:7
        |vpiElemTypespec:
        \_int_typespec: , line:3:13, endln:3:25
          |vpiParent:
          \_array_typespec: , line:3:13, endln:3:52
      |vpiOpType:75
      |vpiOperand:
      \_constant: , line:3:57, endln:3:59
        |vpiParent:
        \_operation: , line:3:55, endln:3:64
        |vpiDecompile:'1
        |vpiSize:32
        |BIN:1
        |vpiConstType:3
      |vpiOperand:
      \_constant: , line:3:61, endln:3:63
        |vpiParent:
        \_operation: , line:3:55, endln:3:64
        |vpiDecompile:'1
        |vpiSize:32
        |BIN:1
        |vpiConstType:3
    |vpiLhs:
    \_parameter: (work@test.NumBobaDrinkers), line:3:26, endln:3:41
```

## Similar test case that works (2)

```systemverilog
module test #(
  parameter int BobaCount = 2,
  parameter int unsigned NumBobaDrinkers[BobaCount] = '{BobaCount{'1}}
)();
endmodule
```

UHDM:
```
|uhdmtopModules:
\_module_inst: work@test (work@test), file:/home/user/p/repos/yosys-systemverilog/test2.sv, line:1:1, endln:5:10
  ...
  |vpiParamAssign:
  \_param_assign: , line:2:17, endln:2:30
    |vpiParent:
    \_module_inst: work@test (work@test), file:/home/user/p/repos/yosys-systemverilog/test2.sv, line:1:1, endln:5:10
    |vpiRhs:
    \_constant: , line:2:29, endln:2:30
      |vpiParent:
      \_param_assign: , line:2:17, endln:2:30
      |vpiDecompile:2
      |vpiSize:32
      |UINT:2
      |vpiTypespec:
      \_int_typespec: , line:2:13, endln:2:16
      |vpiConstType:9
    |vpiLhs:
    \_parameter: (work@test.BobaCount), line:2:17, endln:2:26
  |vpiParamAssign:
  \_param_assign: , line:3:26, endln:3:71
    |vpiParent:
    \_module_inst: work@test (work@test), file:/home/user/p/repos/yosys-systemverilog/test2.sv, line:1:1, endln:5:10
    |vpiRhs:
    \_operation: , line:3:55, endln:3:71
      |vpiParent:
      \_param_assign: , line:3:26, endln:3:71
      |vpiOpType:76
      |vpiOperand:
      \_ref_obj: (work@test.BobaCount), line:3:57, endln:3:66
        |vpiParent:
        \_operation: , line:3:55, endln:3:71
        |vpiName:BobaCount
        |vpiFullName:work@test.BobaCount
        |vpiActual:
        \_parameter: (work@test.BobaCount), line:2:17, endln:2:26
      |vpiOperand:
      \_operation: , line:3:67, endln:3:69
        |vpiParent:
        \_operation: , line:3:55, endln:3:71
        |vpiOpType:33
        |vpiOperand:
        \_constant: , line:3:67, endln:3:69
          |vpiParent:
          \_operation: , line:3:67, endln:3:69
          |vpiDecompile:'1
          |vpiSize:-1
          |BIN:1
          |vpiConstType:3
    |vpiLhs:
    \_parameter: (work@test.NumBobaDrinkers), line:3:26, endln:3:41
```

Ref: https://github.com/chipsalliance/yosys-f4pga-plugins/issues/539

Thomasb81, 2023-08-22:
In this context `'{default: '1}` is "short" hand notation for  `'{'1, '1}`. According to me when a structure initialization is using `default` keyword ; Surelog should elaborate the complete structure initialization depending of the right assignment expression...

But that's going to work only for elaborated data model :  if BobaCount is override with  a value different than 2 since systemVerilog data model does not record `default` information, there will be uninitialized index while source code explicitly specify a default initialization value.

alaindargelas, 2023-08-23:
All the tests generate wrong UHDM. They all need to expand the const to 2^32-1

alaindargelas, 2023-08-23:
#3805 fixes the case:  parameter int unsigned NumBobaDrinkers[BobaCount] = '{'1, '1}

alaindargelas, 2023-08-24:
#3806 fixes the main case: parameter int unsigned NumBobaDrinkers[BobaCount] = '{default: '1}

alaindargelas, 2023-08-25:
#3807 fixes  parameter int unsigned NumBobaDrinkers[BobaCount] = '{BobaCount{'1}}

mglb, 2023-08-29:
There's still an empty typespec:
```
      |vpiTypespec:
      \_array_typespec: , line:3:13, endln:3:52
```
The "Similar test case that works (1)" has full typespec.

alaindargelas, 2023-08-30:
It is not an empty typespec.
Print with the ID: id:12

id:12
```
 \_array_typespec: , id:12, line:3:14, endln:3:53
      |vpiParent:
      \_parameter: (work@test.NumBobaDrinkers), id:6, line:3:27, endln:3:42
      |vpiRange:
      \_range: , id:7, line:3:43, endln:3:52
        |vpiParent:
        \_parameter: (work@test.NumBobaDrinkers), id:6, line:3:27, endln:3:42
        |vpiLeftRange:
        \_constant: , id:8
          |vpiParent:
          \_range: , id:7, line:3:43, endln:3:52
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
          |vpiConstType:9
        |vpiRightRange:
        \_operation: , id:10, line:3:43, endln:3:52
          |vpiParent:
          \_range: , id:7, line:3:43, endln:3:52
          |vpiOpType:11
          |vpiOperand:
          \_ref_obj: (work@test.NumBobaDrinkers.BobaCount), id:9, line:3:43, endln:3:52
            |vpiParent:
            \_operation: , id:10, line:3:43, endln:3:52
            |vpiName:BobaCount
            |vpiFullName:work@test.NumBobaDrinkers.BobaCount
            |vpiActual:
            \_parameter: (work@test.BobaCount), id:35, line:2:18, endln:2:27
          |vpiOperand:
          \_constant: , id:11
            |vpiSize:64
            |INT:1
            |vpiConstType:7
      |vpiElemTypespec:
      \_int_typespec: , id:5, line:3:14, endln:3:26
```

```
====== UHDM =======
design: (work@test), id:32
|vpiElaborated:1
|vpiName:work@test
|uhdmallModules:
\_module_inst: work@test (work@test), id:33, file:/home/alain/Surelog/tests/ScratchPad.sv, line:1:1, endln:5:11
  |vpiParent:
  \_design: (work@test), id:32
  |vpiFullName:work@test
  |vpiParameter:
  \_parameter: (work@test.BobaCount), id:2, line:2:18, endln:2:27
    |vpiParent:
    \_module_inst: work@test (work@test), id:33, file:/home/alain/Surelog/tests/ScratchPad.sv, line:1:1, endln:5:11
    |UINT:2
    |vpiTypespec:
    \_int_typespec: , id:1, line:2:14, endln:2:17
      |vpiParent:
      \_parameter: (work@test.BobaCount), id:2, line:2:18, endln:2:27
      |vpiSigned:1
    |vpiSigned:1
    |vpiName:BobaCount
    |vpiFullName:work@test.BobaCount
  |vpiParameter:
  \_parameter: (work@test.NumBobaDrinkers), id:6, line:3:27, endln:3:42
    |vpiParent:
    \_module_inst: work@test (work@test), id:33, file:/home/alain/Surelog/tests/ScratchPad.sv, line:1:1, endln:5:11
    |vpiSize:1
[... 193 lines cut ...]
    \_parameter: (work@test.BobaCount), id:35, line:2:18, endln:2:27
  |vpiParamAssign:
  \_param_assign: , id:31, line:3:27, endln:3:70
    |vpiParent:
    \_module_inst: work@test (work@test), id:34, file:/home/alain/Surelog/tests/ScratchPad.sv, line:1:1, endln:5:11
    |vpiRhs:
    \_array_expr: , id:27
      |vpiParent:
      \_param_assign: , id:31, line:3:27, endln:3:70
      |vpiTypespec:
      \_array_typespec: , id:12, line:3:14, endln:3:53
      |vpiExpr:
      \_constant: , id:21, line:3:67, endln:3:69
        |vpiDecompile:4294967295
        |vpiSize:32
        |UINT:4294967295
        |vpiConstType:9
      |vpiExpr:
      \_constant: , id:21, line:3:67, endln:3:69
    |vpiLhs:
    \_parameter: (work@test.NumBobaDrinkers), id:37, line:3:27, endln:3:42
  |vpiDefName:work@test
  |vpiTop:1
  |vpiTopModule:1
```

closed 2023-08-30 by alaindargelas

alaindargelas, 2023-08-30:
@mglb can you confirm your are unblocked?