chipsalliance/Surelog #3812: -defer , -link broken when -y used

alaindargelas, 2023-08-29:


alaindargelas, 2023-08-30:
@hs-apotell , I added a test https://github.com/chipsalliance/Surelog/pull/3817/files that showcase that the -defer, -link separate compilation flow is broken with the new filesystem code.
Any path pointing upward: 
Source file:   ../../source.v 
or library:     -y ../../some_path/ 
"corrupts" the FileSystem where files ID point vs where the file is actually in the cache.

cd tests/TestSepComp/badpath
./runsurelog.sh ../../../build/bin/surelog
....
[FTL:PA0202] Cannot open file "/home/alain/Surelog/tests/TestSepComp/badpath/slpp_all/lib/work/badpath/badtop.sv".

alaindargelas, 2023-08-30:
@hs-apotell I see your change which fixes the testcase as a workaround but not the root cause. 
That is not going to work generally. 
Some commands like:    
surelog -defer top.sv -y ../../../lib/   
Would require to be run in two commands:
surelog -defer top.sv 
surelog -defer -wd ../../../  -y lib/

Worse if it's the include path that is relative:
surelog -defer top.sv -I../../../lib/   -I../other/lib/
It is not possible to separate into multiple commands, so some computation has to happen to find the root of all and edit all paths.

Some massive path editing would have to happen meaning we are not compatible with the project definition other tools would accept.
There are some projects running surelog with thousands of files and paths for include and library. They cannot be edited by hand. One of them is at Google and is not working anymore.
Please provide the correct fix.

alaindargelas, 2023-08-30:
@hs-apotell , no all -sepcomp are independent. they all ignore each other.
Only the -link has to know all paths.
Please find a solution where:
 - no path has to be edited from original command line (Standard Verilog command line accepted by other tools - Commercial simulators) when making the -sepcomp calls.
- The -link command has to find all the information from the <hash>.lst files created by the -sepcomp. The command line should not require users to add -wd .... arguments.

alaindargelas, 2023-08-30:
Basically we need to restore compatibility with the Verilog Standard  path for .v -y -I -v arguments and accept the project definition as supported by other tools (Commercial simulators)

QuantamHD, 2023-08-30:
Not sure if it makes a difference, but when we compile on distributed cluster at Google only the relative paths are preserved each invocation of surelog is run with a new root directory.

Job 1 
/tmp/job-iejfowiejfoiwjeoif/hardware/verilog.v

Job 2
/tmp/job-ssseiieieuiejej/hardware/verilog.v
/tmp/job-ssseiieieuiejej/hardware/verilog2.v

alaindargelas, 2023-08-30:
@hs-apotell , this week-end is fine.
Google and I are blocked by this, but the work around is not to use -sepcomp, -link in the meantime.
Note that I have a more complex setup than the one @QuantamHD mentions, the projects I need to run have a lot of relative paths, going any direction, upward, downward and absolute paths pointing to other places than the current working dir.

QuantamHD, 2023-08-30:
The link command occurs on a single machine with a single working directory.

So when running link the files would be available here

/tmp/job-final/verilog1.v
/tmp/job-final/verilog2.v
/tmp/job-final/verilog3.v
/tmp/job-final/verilog4.v
/tmp/job-final/verilog5.v

QuantamHD, 2023-08-30:
@hs-apotell Ah the working directory is always at /tmp/job-final/