chipsalliance/Surelog #3849: Logging: "Creating log file" would be nicer without full-stop to allow for common terminal copy/paste workflow

hzeller, 2023-09-11:
Surelog emits log messages that tell which logfile is created. These messages have a full-stop at the end like other log messages:

```
[INF:CM0023] Creating log file foo/bar/slpp_all/surelog.log.
```

The typical use-case would be to double-click the filename in a terminal to copy it and paste somewhere else; but then, the filename has a trailing dot. So right now, the way the log message is formatted requires the user to remove the dot every time they copy/paste.

So a usability improvement would be if there was _no_ dot after a logged filename.

hzeller, 2023-09-11:
This could probably simple work-arounded by putting the file-name in quotes
```
[INF:CM0023] Creating log file "foo/bar/slpp_all/surelog.log".
```

alaindargelas, 2023-09-28:
Fixed by #3890

closed 2023-09-28 by alaindargelas