chipsalliance/Surelog #3854: Clean install - other projects have cmake issues with find_package

pieter3d, 2023-09-13:
This is after a from-scratch repo clone, build and install. No additional make switches.

 I tried building a different project that contains `find_package(Surelog REQUIRED)` in CmakeLists.txt. Cmake complains as follows:
```
CMake Error at /usr/local/cmake/Surelog.cmake:118 (message):
  The imported target "surelog" references the file

     "/usr/local/lib/surelog/libsurelog.a"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/local/cmake/Surelog.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/local/cmake/SurelogConfig.cmake:10 (include)
  src/CMakeLists.txt:10 (find_package)
```

The file `/usr/local/lib/surelog/libsurelog.a` indeed doesn't exist, but `/usr/local/lib/libsurelog.a` does. So there's some prefix being assumed somewhere...

pieter3d, 2023-09-13:
Seems to have something to do with https://github.com/chipsalliance/Surelog/pull/3828/commits/ca42785a0502e3e90ad677695991a74ff4876fdf

pieter3d, 2023-09-13:
Ah - nevermind. An old install of Surelog wasn't cleaned up properly, there was stuff in `/usr/local/cmake/Surelog*`

closed 2023-09-13 by pieter3d

timkpaine, 2023-09-13:
I too got tired of manually installing/uninstalling things, homebrew is in progress, rpms in progress, debian files here: https://github.com/dau-dev/tools/releases/tag/v0.0.1.