chipsalliance/Surelog #3917: Unresolved hierarchical reference calculating $bits() of a unpacked member

hzeller, 2023-10-30:
The following

```SystemVerilog
typedef struct packed {
   logic [31:0] pair[2];
} foo_t;

module a(input foo_t a);
   // Getting size of full pair: works
   logic [$bits(a.pair)-1:0] bar;

   // but referencing one element: unresolved hierarchical reference
   logic [$bits(a.pair[0])-1:0] baz;  // <-- problem here
endmodule;
```

.. results in an error:

```
file.sv:10: Unresolved hierarchical reference "a.pair[0]".
```

This must've been recent; with [Surelog v.1.76](https://github.com/chipsalliance/Surelog/releases/tag/v1.76), this compiled.

alaindargelas, 2023-10-30:
The testcase is illegal as-is.
You have to drop the "packed" keyword.
A packed struct cannot contain unpacked data members.

alaindargelas, 2023-10-30:
Fixed by #3918

closed 2023-10-30 by alaindargelas

alaindargelas, 2023-10-31:
Bumped version 1.77

Thomasb81, 2023-10-31:
> A packed struct cannot contain unpacked data members.

I am interested by an LRM reference.

hzeller, 2023-10-31:
7.4.1 Packed arrays (4th paragraph)
_"Packed arrays can be made of only the single bit data types (bit, logic, reg ), enumerated types, and
recursively other packed arrays and **packed structures**."_

hzeller, 2023-10-31:
In response to https://github.com/chipsalliance/Surelog/issues/3917#issuecomment-1787718044 this is the relevant section:

7.2.1 
"..._Only packed data types and the integer data types summarized in Table 6-8 (see 6.11) shall be legal in
packed structures_"

Thomasb81, 2023-10-31:
@hzeller Tanks a lot.