chipsalliance/Surelog #3958: Connect-by-name has no vpiHighConn

pieter3d, 2024-01-28:
```verilog

my_module module sub (
  input logic clk,
  input logic data_in,
  output logic data_out
);
endmodule

module test;

logic clk;
logic data_in_wire;
sub sub_i (
  .clk,
  .data_in (data_in_wire),
  .data_out (/* no connect */)
);

endmodule 
````

Dumping the UHDM for that shows this (instance only):
```
  |vpiRefModule:
  \_ref_module: work@sub (sub_i), line:12:5, endln:12:10
    |vpiParent:
    \_module_inst: work@test (work@test), file:/home/pieter/work/pieter3d/simview/build/test.sv, line:8:1, endln:18:10
    |vpiName:sub_i
    |vpiDefName:work@sub
    |vpiActual:
    \_module_inst: work@sub (work@sub), file:/home/pieter/work/pieter3d/simview/build/test.sv, line:1:1, endln:6:10
    |vpiPort:
    \_port: (clk), line:13:3, endln:13:7
      |vpiParent:
      \_ref_module: work@sub (sub_i), line:12:5, endln:12:10
      |vpiName:clk
    |vpiPort:
    \_port: (data_in), line:14:3, endln:14:26
      |vpiParent:
      \_ref_module: work@sub (sub_i), line:12:5, endln:12:10
      |vpiName:data_in
      |vpiHighConn:
      \_ref_obj: (work@test.sub_i.data_in.data_in_wire), line:14:13, endln:14:25
        |vpiParent:
        \_port: (data_in), line:14:3, endln:14:26
        |vpiName:data_in_wire
        |vpiFullName:work@test.sub_i.data_in.data_in_wire
        |vpiActual:
        \_logic_net: (work@test.data_in_wire), line:11:7, endln:11:19
    |vpiPort:
    \_port: , line:15:3, endln:15:31
      |vpiParent:
      \_ref_module: work@sub (sub_i), line:12:5, endln:12:10

```

So at least two things look wrong to me:
* There appears to be no connection to `clk` but it's a valid connect-by-name contstruct
* The floating `data_out` output is totally missing, the port has a blank name.

alaindargelas, 2024-01-31:
[pieter3d](https://github.com/pieter3d) this is working just fine:

```
|uhdmtopModules:
\_module_inst: work@test (work@test), id:61, file:/home/alain/Surelog/tests/ScratchPad.sv, line:8:1, endln:18:10
  |vpiName:work@test
  |vpiVariables:
  \_logic_var: (work@test.clk), id:7, line:10:7, endln:10:10
    |vpiParent:
    \_module_inst: work@test (work@test), id:61, file:/home/alain/Surelog/tests/ScratchPad.sv, line:8:1, endln:18:10
    |vpiTypespec:
    \_ref_typespec: (work@test.clk), id:8
      |vpiParent:
      \_logic_var: (work@test.clk), id:7, line:10:7, endln:10:10
      |vpiFullName:work@test.clk
      |vpiActual:
      \_logic_typespec: , id:6, line:10:1, endln:10:6
    |vpiName:clk
    |vpiFullName:work@test.clk
    |vpiVisibility:1
  |vpiVariables:
  \_logic_var: (work@test.data_in_wire), id:10, line:11:7, endln:11:19
    |vpiParent:
    \_module_inst: work@test (work@test), id:61, file:/home/alain/Surelog/tests/ScratchPad.sv, line:8:1, endln:18:10
    |vpiTypespec:
    \_ref_typespec: (work@test.data_in_wire), id:11
      |vpiParent:
      \_logic_var: (work@test.data_in_wire), id:10, line:11:7, endln:11:19
      |vpiFullName:work@test.data_in_wire
      |vpiActual:
      \_logic_typespec: , id:9, line:11:1, endln:11:6
    |vpiName:data_in_wire
    |vpiFullName:work@test.data_in_wire
    |vpiVisibility:1
  |vpiDefName:work@test
  |vpiTop:1
  |vpiTopModule:1
  |vpiModule:
  \_module_inst: work@sub (work@test.sub_i), id:62, file:/home/alain/Surelog/tests/ScratchPad.sv, line:12:1, endln:16:3
    |vpiParent:
    \_module_inst: work@test (work@test), id:61, file:/home/alain/Surelog/tests/ScratchPad.sv, line:8:1, endln:18:10
    |vpiName:sub_i
    |vpiFullName:work@test.sub_i
    |vpiDefName:work@sub
    |vpiDefFile:/home/alain/Surelog/tests/ScratchPad.sv
    |vpiDefLineNo:1
    |vpiNet:
    \_logic_net: (work@test.sub_i.clk), id:22, line:2:15, endln:2:18
      |vpiParent:
      \_module_inst: work@sub (work@test.sub_i), id:62, file:/home/alain/Surelog/tests/ScratchPad.sv, line:12:1, endln:16:3
      |vpiTypespec:
      \_ref_typespec: (work@test.sub_i.clk), id:23
        |vpiParent:
        \_logic_net: (work@test.sub_i.clk), id:22, line:2:15, endln:2:18
        |vpiFullName:work@test.sub_i.clk
        |vpiActual:
        \_logic_typespec: , id:21, line:2:9, endln:2:14
      |vpiName:clk
      |vpiFullName:work@test.sub_i.clk
      |vpiNetType:36
    |vpiNet:
    \_logic_net: (work@test.sub_i.data_in), id:25, line:3:15, endln:3:22
      |vpiParent:
      \_module_inst: work@sub (work@test.sub_i), id:62, file:/home/alain/Surelog/tests/ScratchPad.sv, line:12:1, endln:16:3
      |vpiTypespec:
      \_ref_typespec: (work@test.sub_i.data_in), id:26
        |vpiParent:
        \_logic_net: (work@test.sub_i.data_in), id:25, line:3:15, endln:3:22
        |vpiFullName:work@test.sub_i.data_in
        |vpiActual:
[... 86 lines cut ...]
      |vpiName:data_out
      |vpiDirection:2
      |vpiHighConn:
      \_operation: , id:72, line:15:30, endln:15:31
        |vpiParent:
        \_port: (data_out), id:71, line:4:16, endln:4:24
        |vpiOpType:36
      |vpiLowConn:
      \_ref_obj: (work@test.sub_i.data_out), id:73, line:4:16, endln:4:24
        |vpiParent:
        \_port: (data_out), id:71, line:4:16, endln:4:24
        |vpiName:data_out
        |vpiFullName:work@test.sub_i.data_out
        |vpiActual:
        \_logic_net: (work@test.sub_i.data_out), id:28, line:4:16, endln:4:24
      |vpiTypedef:
      \_ref_typespec: (work@test.sub_i.data_out), id:74
        |vpiParent:
        \_port: (data_out), id:71, line:4:16, endln:4:24
        |vpiFullName:work@test.sub_i.data_out
        |vpiActual:
        \_logic_typespec: , id:19, line:4:10, endln:4:15
      |vpiInstance:
      \_module_inst: work@sub (work@test.sub_i), id:62, file:/home/alain/Surelog/tests/ScratchPad.sv, line:12:1, endln:16:3
```

alaindargelas, 2024-01-31:
You need to look at the elaborated tree

pieter3d, 2024-02-02:
Are you sure that's right? the port "clk" has both the the HighConn and LowConn vpiActual as the inner net of the sub module - `work@test.sub_i.clk`. But data_in has HighConn as `work@test.data_in_wire` (the net in the top) and LowConn as `work@test.sub_i.data_in`, the net in the submodule.

pieter3d, 2024-02-02:
It's more obvious when the same module is connected different ways (all logically equivalent):

```systemverilog
module sub (
  input logic clk,
);
endmodule

module test;

logic clk;

sub sub0 ( .* );
sub sub1 ( .clk );
sub sub2 ( .clk (clk) );

endmodule 
```

pieter3d, 2024-02-06:
UHDM dump of the above:
```
|uhdmtopModules:
\_module_inst: work@test (work@test), file:/home/pieter/work/pieter3d/Surelog/dbuild/bin/test.sv, line:6:1, endln:16:10
  |vpiName:work@test
  |vpiVariables:
  \_logic_var: (work@test.clk), line:8:7, endln:8:10
    |vpiParent:
    \_module_inst: work@test (work@test), file:/home/pieter/work/pieter3d/Surelog/dbuild/bin/test.sv, line:6:1, endln:16:10
    |vpiTypespec:
    \_ref_typespec: (work@test.clk)
      |vpiParent:
      \_logic_var: (work@test.clk), line:8:7, endln:8:10
      |vpiFullName:work@test.clk
      |vpiActual:
      \_logic_typespec: , line:8:1, endln:8:6
    |vpiName:clk
    |vpiFullName:work@test.clk
    |vpiVisibility:1
  |vpiVariables:
  \_logic_var: (work@test.clk_wire), line:9:7, endln:9:15
    |vpiParent:
    \_module_inst: work@test (work@test), file:/home/pieter/work/pieter3d/Surelog/dbuild/bin/test.sv, line:6:1, endln:16:10
    |vpiTypespec:
    \_ref_typespec: (work@test.clk_wire)
      |vpiParent:
      \_logic_var: (work@test.clk_wire), line:9:7, endln:9:15
      |vpiFullName:work@test.clk_wire
      |vpiActual:
      \_logic_typespec: , line:9:1, endln:9:6
    |vpiName:clk_wire
    |vpiFullName:work@test.clk_wire
    |vpiVisibility:1
  |vpiDefName:work@test
  |vpiTop:1
  |vpiTopModule:1
  |vpiModule:
  \_module_inst: work@sub (work@test.sub0), file:/home/pieter/work/pieter3d/Surelog/dbuild/bin/test.sv, line:11:1, endln:11:17
    |vpiParent:
    \_module_inst: work@test (work@test), file:/home/pieter/work/pieter3d/Surelog/dbuild/bin/test.sv, line:6:1, endln:16:10
    |vpiName:sub0
    |vpiFullName:work@test.sub0
    |vpiDefName:work@sub
    |vpiDefFile:/home/pieter/work/pieter3d/Surelog/dbuild/bin/test.sv
    |vpiDefLineNo:1
    |vpiNet:
    \_logic_net: (work@test.sub0.clk), line:2:15, endln:2:18
      |vpiParent:
      \_module_inst: work@sub (work@test.sub0), file:/home/pieter/work/pieter3d/Surelog/dbuild/bin/test.sv, line:11:1, endln:11:17
      |vpiTypespec:
      \_ref_typespec: (work@test.sub0.clk)
        |vpiParent:
        \_logic_net: (work@test.sub0.clk), line:2:15, endln:2:18
        |vpiFullName:work@test.sub0.clk
        |vpiActual:
        \_logic_typespec: , line:2:9, endln:2:14
      |vpiName:clk
      |vpiFullName:work@test.sub0.clk
      |vpiNetType:36
    |vpiInstance:
    \_module_inst: work@test (work@test), file:/home/pieter/work/pieter3d/Surelog/dbuild/bin/test.sv, line:6:1, endln:16:10
    |vpiPort:
    \_port: (clk), line:2:15, endln:2:18
      |vpiParent:
      \_module_inst: work@sub (work@test.sub0), file:/home/pieter/work/pieter3d/Surelog/dbuild/bin/test.sv, line:11:1, endln:11:17
      |vpiName:clk
      |vpiDirection:1
      |vpiHighConn:
      \_ref_obj: (work@test.clk), line:11:12, endln:11:13
        |vpiParent:
[... 166 lines cut ...]
      \_ref_obj: (work@test.clk_wire), line:14:18, endln:14:26
        |vpiParent:
        \_port: (clk), line:2:15, endln:2:18
        |vpiName:clk_wire
        |vpiFullName:work@test.clk_wire
        |vpiActual:
        \_logic_var: (work@test.clk_wire), line:9:7, endln:9:15
      |vpiLowConn:
      \_ref_obj: (work@test.sub3.clk), line:14:13, endln:14:16
        |vpiParent:
        \_port: (clk), line:2:15, endln:2:18
        |vpiName:clk
        |vpiFullName:work@test.sub3.clk
        |vpiActual:
        \_logic_net: (work@test.sub3.clk), line:2:15, endln:2:18
      |vpiTypedef:
      \_ref_typespec: (work@test.sub3.clk)
        |vpiParent:
        \_port: (clk), line:2:15, endln:2:18
        |vpiFullName:work@test.sub3.clk
        |vpiActual:
        \_logic_typespec: , line:2:9, endln:2:14
      |vpiInstance:
      \_module_inst: work@sub (work@test.sub3), file:/home/pieter/work/pieter3d/Surelog/dbuild/bin/test.sv, line:14:1, endln:14:30
```

pieter3d, 2024-02-06:
I poked around some in NetlistElaboration.cpp, thanks for the suggestion. At the point where `sub0` is being elaborated, it actually does properly set High_conn to an `obj_ref` who's `Actual_group` is the net inside `test`, and not the net inside `sub0` like the uhdm dump shows. But then after elaboration is done, doing some quick and dirty inspections like this in `main.cc`:
```c++
  UHDM::any* con = 
         ((UHDM::design *)((uhdm_handle *)vpi_design)->object)
                         ->TopModules()
                         ->at(0)
                         ->Modules() // name is "sub0"
                         ->at(0)
                         ->Ports()->at(0)->High_conn();
  UHDM::ref_obj* o = (UHDM::ref_obj*)con;
  printf("debug: %d %d %s\n", 
      o->Actual_group()->VpiType(),
      o->Actual_group()->VpiLineNo(),
      std::string(o->Actual_group()->VpiName()).c_str());
```
shows the `Actual_group` pointing back to the internal net, like the UHDM dump.

So either the elaborated module gets replaced, or I am not looking at the right elaborated tree somehow. I don't think the High_conn itself is getting overwritten later, based on some debug prints and watching addresses in a debugger, but I am not 100% sure.

So - not quite sure yet where the issues lies.

pieter3d, 2024-02-07:
Ok, making some progress. The High_conn's `Actual_group` is getting overridden when UHDM is being written out. This is either a bug in UHDM, or a bug in how it's being used.

Culprit line: https://github.com/chipsalliance/UHDM/blob/b918b5d5c82cfbaf71e04f87ab69fc7e182e41aa/templates/ElaboratorListener.cpp#L907

During write-out, if the net name matches any local nets, it gets replaced.

alaindargelas, 2024-02-09:
The instStack_ has to be popped temporarily while binding high_conn so is skips the current module scope.

pieter3d, 2024-02-09:
Yeah that's kinda what I figured, but the Listener classes need some new state to know to do that, i.e. when they are processing a High_conn

alaindargelas, 2024-02-09:
Set a bp in the bind function, check the stack, you should see the high conn relation traversal, the python code generator in scripts/ needs to change to pop and push back the context before and after the traversal of that particular relation

pieter3d, 2024-02-09:
Yes, but that's all c++ code... Where does python come into this?

alaindargelas, 2024-02-09:
Most of the C++ code is auto generated by Python