chipsalliance/Surelog #3959: Can't determine struct member size in elaborated designs

pieter3d, 2024-01-28:
Consider the following design:
```systemverilog
module test;

typedef struct packed {
  logic [7:0] a;
  logic [7:0] b;
} pair_t;

pair_t pair;

endmodule
```
UHDM dump:
```
  \_struct_typespec: (pair_t), line:3:9, endln:6:2
    |vpiParent:
    \_module_inst: work@test (work@test), file:/home/pieter/work/pieter3d/simview/build/struct_test.sv, line:1:1, endln:10:10
    |vpiName:pair_t
    |vpiInstance:
    \_module_inst: work@test (work@test), file:/home/pieter/work/pieter3d/simview/build/struct_test.sv, line:1:1, endln:10:10
    |vpiPacked:1
    |vpiTypespecMember:
    \_typespec_member: (a), line:4:15, endln:4:16
      |vpiParent:
      \_struct_typespec: (pair_t), line:3:9, endln:6:2
      |vpiName:a
      |vpiTypespec:
      \_ref_typespec: (work@test.pair_t.a)
        |vpiParent:
        \_typespec_member: (a), line:4:15, endln:4:16
        |vpiFullName:work@test.pair_t.a
        |vpiActual:
        \_logic_typespec: , line:4:3, endln:4:14
      |vpiRefFile:/home/pieter/work/pieter3d/simview/build/struct_test.sv
      |vpiRefLineNo:4
      |vpiRefColumnNo:3
      |vpiRefEndLineNo:4
      |vpiRefEndColumnNo:14
    |vpiTypespecMember:
    \_typespec_member: (b), line:5:15, endln:5:16
      |vpiParent:
      \_struct_typespec: (pair_t), line:3:9, endln:6:2
      |vpiName:b
      |vpiTypespec:
      \_ref_typespec: (work@test.pair_t.b)
        |vpiParent:
        \_typespec_member: (b), line:5:15, endln:5:16
        |vpiFullName:work@test.pair_t.b
        |vpiActual:
        \_logic_typespec: , line:5:3, endln:5:14
      |vpiRefFile:/home/pieter/work/pieter3d/simview/build/struct_test.sv
      |vpiRefLineNo:5
      |vpiRefColumnNo:3
      |vpiRefEndLineNo:5
      |vpiRefEndColumnNo:14
```
The typespec_members are correctly called out as type logic, but there is no further information, no way to determine the width or other structure of the member.

alaindargelas, 2024-01-31:
You need to use the UHDM::ExprEval::size  while traversing the elaborated model (topModules):
https://github.com/chipsalliance/UHDM/blob/b918b5d5c82cfbaf71e04f87ab69fc7e182e41aa/templates/ExprEval.h#L60

alaindargelas, 2024-01-31:
You also can traverse any member (The code of ExprEval::size does just that):
https://github.com/chipsalliance/UHDM/blob/b918b5d5c82cfbaf71e04f87ab69fc7e182e41aa/templates/ExprEval.cpp#L1407

closed 2024-01-31 by alaindargelas