chipsalliance/Surelog #3960: Wrong Parameter calculation in the elaborated model

Pietro4F, 2024-02-06:
Hello,
I've notice an error in the calculation of a parameter in the elaborated model.

I have different files with different modules, one of them is similar to this...

```verilog
module abc #(
    // --
    // Parameters
    // --
    parameter A = 8_000_000,
    parameter B = 100,
    ...
    )
    (
    // --
    // Clock and ...
    // --
    input wire                       clk,
    input wire                      .......
 );

 localparam C= B * (A / 1_000_000);
 
```

In the not elaborated tree i have this

```
|vpiParamAssign:
\_param_assign: , line:x, endln:x
  |vpiParent:
  \_module_inst: work@abc (work@abc), file:abc.v, line:x, endln:x
  |vpiRhs:
  \_constant: , line:x, endln:x
    |vpiParent:
    \_param_assign: , line:x, endln:x
    |vpiDecompile:8_000_000
    |vpiSize:64
    |UINT:8
    |vpiTypespec:
    \_ref_typespec: (work@abc)
      |vpiParent:
      \_constant: , line:x, endln:x
      |vpiFullName:work@abc
      |vpiActual:
      \_int_typespec: , line:x, endln:x
    |vpiConstType:9
  |vpiLhs:
  \_parameter: (work@abc.A), line:x, endln:x
.
.
.
|vpiParamAssign:
\_param_assign: , line:x, endln:x
  |vpiParent:
  \_module_inst: work@abc (work@abc), file:abc.v, line:x, endln:x
  |vpiRhs:
  \_constant: , line:x, endln:x
    |vpiParent:
    \_param_assign: , line:x, endln:x
    |vpiDecompile:100
    |vpiSize:64
    |UINT:100
    |vpiTypespec:
    \_ref_typespec: (work@abc)
      |vpiParent:
      \_constant: , line:x, endln:x
      |vpiFullName:work@abc
      |vpiActual:
      \_int_typespec: , line:x, endln:x
    |vpiConstType:9
  |vpiLhs:
  \_parameter: (work@abc.B), line:x, endln:x
.
.
.
|vpiParamAssign:
  \_param_assign: , line:x, endln:x
    |vpiParent:
    \_module_inst: work@abc (work@abc), file:abc.v, line:x, endln:x
    |vpiRhs:
    \_operation: , line:x, endln:x
      |vpiParent:
      \_param_assign: , line:x, endln:x
      |vpiOpType:25
      |vpiOperand:
      \_ref_obj: (work@abc.B), line:x, endln:x
        |vpiParent:
        \_operation: , line:x, endln:x
        |vpiName:B
        |vpiFullName:work@abc.B
      |vpiOperand:
      \_operation: , line:x, endln:x
        |vpiParent:
        \_operation: , line:x, endln:x
        |vpiOpType:12
        |vpiOperand:
        \_ref_obj: (work@abc.A), line:x, endln:x
          |vpiParent:
          \_operation: , line:x, endln:x
          |vpiName:A
          |vpiFullName:work@abc.A
        |vpiOperand:
        \_constant: , line:x, endln:x
          |vpiParent:
          \_operation: , line:x, endln:x
          |vpiDecompile:1_000_000
          |vpiSize:64
          |UINT:1
          |vpiConstType:9
    |vpiLhs:
    \_parameter: (work@abc.C), line:x, endln:x
```

But the in the elaborated tree i obtain 800000000 instead 800.

```

    |vpiParamAssign:
    \_param_assign: , line:x, endln:x
        |vpiParent:
        \_module_inst: work@abc (work@module1.module2.module3.module4.abc), file:abc.v, line:x, endln:x
        |vpiRhs:
        \_constant: , line:x, endln:x
        |vpiParent:
        \_param_assign: , line:x, endln:x
        |vpiDecompile:800000000
        |vpiSize:64
        |INT:800000000
        |vpiConstType:7
        |vpiLhs:
        \_parameter: (work@module1.module2.module3.module4.abc.C), line:x, endln:x

```
During the inclusion a give to the parameters the same values ad the default ones.

I have tried the same thing but with one simple module in one file.
So my module in that case is not included anywhere and I can obtain the correct result 800.

alaindargelas, 2024-02-09:
@Pietro4F , please add a test that is a complete Verilog file (That at least parses and compile, even if it is minimal). Not just snippet of code.

Pietro4F, 2024-02-09:
I think the problem is just the it can't recognize the number with **_** , for example It probably counts the 1_000_000 in the division as just 1, the same for 8_000_000.
I try to recreate the same error.

``` SystemVerilog
module abc #(
  parameter A = 8_000_000,
  parameter B = 100
)(
  input wire clk,
  input wire rst
);

  localparam C = B * (A / 1_000_000);

endmodule

module top (
  input wire clk,
  input wire rst
);

  abc #( .A(8_000_000), .B(100) ) u_abc (
    .clk(clk),
    .rst(rst)
  );

endmodule
```

In this first case C has the correct value only because A has the value of 8.

```

|uhdmtopModules:
\_module_inst: work@top (work@top), file:/home/top.v, line:13:1, endln:23:10
  |vpiName:work@top
  |vpiDefName:work@top
  |vpiTop:1
  |vpiNet:
  \_logic_net: (work@top.clk), line:14:14, endln:14:17
    |vpiParent:
    \_module_inst: work@top (work@top), file:/home/top.v, line:13:1, endln:23:10
    |vpiTypespec:
    \_ref_typespec: (work@top.clk)
      |vpiParent:
      \_logic_net: (work@top.clk), line:14:14, endln:14:17
      |vpiFullName:work@top.clk
      |vpiActual:
      \_logic_typespec: , line:14:9, endln:14:13
    |vpiName:clk
    |vpiFullName:work@top.clk
    |vpiNetType:1
  |vpiNet:
  \_logic_net: (work@top.rst), line:15:14, endln:15:17
    |vpiParent:
    \_module_inst: work@top (work@top), file:/home/top.v, line:13:1, endln:23:10
    |vpiTypespec:
    \_ref_typespec: (work@top.rst)
      |vpiParent:
      \_logic_net: (work@top.rst), line:15:14, endln:15:17
      |vpiFullName:work@top.rst
      |vpiActual:
      \_logic_typespec: , line:15:9, endln:15:13
    |vpiName:rst
    |vpiFullName:work@top.rst
    |vpiNetType:1
  |vpiTopModule:1
  |vpiPort:
  \_port: (clk), line:14:14, endln:14:17
    |vpiParent:
[... 329 lines cut ...]
        \_ref_typespec: (work@top.u_abc)
          |vpiParent:
          \_constant: , line:3:17, endln:3:20
          |vpiFullName:work@top.u_abc
          |vpiActual:
          \_int_typespec: , line:3:3, endln:3:20
        |vpiConstType:9
      |vpiLhs:
      \_parameter: (work@top.u_abc.B), line:3:13, endln:3:14
    |vpiParamAssign:
    \_param_assign: , line:9:14, endln:9:37
      |vpiParent:
      \_module_inst: work@abc (work@top.u_abc), file:/home/top.v, line:18:3, endln:21:5
      |vpiRhs:
      \_constant: , line:9:18, endln:9:37
        |vpiParent:
        \_param_assign: , line:9:14, endln:9:37
        |vpiDecompile:800000000
        |vpiSize:64
        |INT:800000000
        |vpiConstType:7
      |vpiLhs:
      \_parameter: (work@top.u_abc.C), line:9:14, endln:9:15

```

alaindargelas, 2024-02-22:
Fixed in #3962

closed 2024-02-22 by alaindargelas