chipsalliance/Surelog #3973: Getting ports in a module

mysoreanoop, 2024-05-27:
I'm trying to get ports of a module with the vpiHandle of the module like so:
```
if(vpiHandle ports = vpi_iterate(vpiPorts, m)) {     // m is the handle for the module currently in
  std::cout << "Found ports\n";
  // do something with it
} else std::cout << "No ports found\n";
```
This never seems to be finding any ports.

I've done this with vpiContAssign, vpiProcess, etc. which all work as expected. I see similar implementations for all these in uhdm/generated/src/module_inst.cpp

alaindargelas, 2024-06-02:
Start with working examples:

https://github.com/chipsalliance/UHDM/blob/master/util/uhdm-hier.cpp

https://github.com/chipsalliance/synlig/blob/main/frontends/systemverilog/uhdm_ast.cc#L2321

https://github.com/chipsalliance/Surelog/blob/00df97c73df40ba9633055f7a166a64421915021/src/hellouhdm.cpp#L158

mysoreanoop, 2024-06-24:
Thanks, I figured it out.