chipsalliance/Surelog #3974: Missing functional cover items in UHDM model

dippmatt, 2024-06-04:
I am trying to understand Surelogs capability to parse Testbench / Verification environment components, specifically, functional coverage. For that reason I assembled a very simple module including a Covergroup and Coverpoint, that is filled during simulation attatched below.

The goal in this case is to extract the statically compilable functional coverage model from a testbench (all coverpoints, assertions,..).

In the UHDM dump of the attatched module, there is no information about the coverpoint and the created bins (min_vals, med_vals, max_vals), I only find function calls to the 'sample' function, e.g. `_method_func_call: (sample), line:23:10, endln:23:16`. I would have expected a class definition of the covergroup, including the bins as well.

Is this expected / intended behaviour of Surelog? If so I'd be interested why this behaviour is observed.

```systemverilog
`ifndef __COVERGROUP__
`define __COVERGROUP__

module covergroup_test ();
   covergroup covergroup_data (string name) 
   with function sample (logic [15:0] data);
      option.per_instance = 1;
      option.name         = name;

      data_cp: coverpoint (data) {
	 bins min_vals[] = {16'h0};
	 bins med_vals[] = {[16'h1:16'hfffe]};
	 bins max_vals[] = {16'hffff};
      }
   endgroup : covergroup_data
   
   covergroup_data cg = new("cg");

   initial begin : test
      #1ns;
      cg.sample(16'h0);
      #1ns;
      cg.sample(16'hff);
      #1ns;
      cg.sample(16'hffff);
      $finish();
   end : test
   
endmodule : covergroup_test
   
`endif // __COVERGROUP__
```

Thomasb81, 2024-06-04:
Hello

For the time being, Surelog has been focus on design aspect. Verification construction are more incomplete.
Additionally the VPI object model, even in latest systemVerilog 2023 revision does not explicitly discuss covergroup construction.

Currently in the uhdm dump the covergroup is reported as :
```
      \_unsupported_typespec: (covergroup_data), line:18:4, endln:18:34
```
Which let think that the construction fall in a fallback because not implemented.

Probably that Surelog should fill the uhdm with a class and required attribute as an user would implement it without this specific construction.

dippmatt, 2024-06-05:
Hello @Thomasb81,

 thanks for the clarification!

closed 2024-06-05 by dippmatt