chipsalliance/Surelog #3975: Efficient backtracking from output ports to input ports

mysoreanoop, 2024-06-24:
I'm trying to backtrack output ports from the topModule this way:
1a. Parse all continuous and procedural assignments.
1b. For each assignment, populate an lhs2rhs map (vpiFullName is used as unique ID).
2. Parse all submodule instances for IOs: form a lowConn to highConn map called io2net map.
3. Starting at the topModule, get the ports, vpiDirection, and if it's an output port:
4a. Search lhs2rhs, io2net map:
4b. If found, recurse to 4a
4c. Else if topModule's input port is reached, end.
4d. Else if constant or parameter, end.
4e. Else mapping not found / other cases.

Is there a more efficient way to do this without having to parse all assignments and submodule IOs?
Specifically, given a vpiHandle to an output port, could I get all the sources of that port/net and keep doing that recursively?
Would procedural assignments with if-else branches complicate things?

alaindargelas, 2024-06-25:
@mysoreanoop do you intent to contribute your code to main? (Hint: If you don't, no help :-))
If you intend to, then it should be a new utility class in UHDM. Let me know, and I'll help you to shape it so it more generic.

mysoreanoop, 2024-06-25:
I'd be happy to contribute back. I can get started with a preliminary PR and we can build on it, or I can develop on it separately and pull when ready; your call.

Thomasb81, 2024-06-25:
Hello

Is it a step forward to solve : https://github.com/chipsalliance/UHDM/issues/1037
Or a different feature ?

alaindargelas, 2024-06-25:
@Thomasb81, It is indeed a step toward: https://github.com/chipsalliance/UHDM/issues/1037
@mysoreanoop, PRs and this discussions should move to the UHDM repo, switching to:
https://github.com/chipsalliance/UHDM/issues/1078