chipsalliance/Surelog #3999: Incorrect complement of partly filled enums

Szelwiga, 2024-10-29:
## Incorrect complement of partly filled enums

Parsing SystemVerilog code (a.sv):
```
typedef enum logic [1:0] {
    A = 3, B = 0, C, D
} state_t;

module top(input clk, output z);
    state_t state = A;

    always @(posedge clk) begin
        case (state)
            A: state <= B;
            B: state <= C;
            C: state <= D;
            default: state <= A;
        endcase
    end

    assign z = (state == B);
endmodule
```
with `surelog -parse -elabuhdm a.sv`, results in UHDM Tree that sets both `enum` values `A` and `D` to `3`. Below is UHDM Tree fragment responsible for `enum`.

```
  |vpiEnumConst:
  \_enum_const: (A), line:2:5, endln:2:10
    |vpiParent:
    \_enum_typespec: (state_t), line:1:1, endln:3:11
    |vpiName:A
    |UINT:3
    |vpiDecompile:3
    |vpiSize:64
  |vpiEnumConst:
  \_enum_const: (B), line:2:12, endln:2:17
    |vpiParent:
    \_enum_typespec: (state_t), line:1:1, endln:3:11
    |vpiName:B
    |UINT:0
    |vpiDecompile:0
    |vpiSize:64
  |vpiEnumConst:
  \_enum_const: (C), line:2:19, endln:2:20
    |vpiParent:
    \_enum_typespec: (state_t), line:1:1, endln:3:11
    |vpiName:C
    |INT:2
    |vpiDecompile:2
    |vpiSize:2
  |vpiEnumConst:
  \_enum_const: (D), line:2:22, endln:2:23
    |vpiParent:
    \_enum_typespec: (state_t), line:1:1, endln:3:11
    |vpiName:D
    |INT:3
    |vpiDecompile:3
    |vpiSize:2
```

alaindargelas, 2025-05-07:
@Szelwiga what does the LRM says in this case?

alaindargelas, 2025-06-18:
The values can be cast to integer types and increment from an initial value of 0. This can be overridden.
enum {bronze=3, silver, gold} medal; // silver=4, gold=5
The values can be set for some of the names and not set for other names. The optional value of an enum
named constant is an elaboration time constant expression (see 6.20) and can include references to
parameters, local parameters, genvars, other enum named constants, and constant functions of these.
Hierarchical names and const variables are not allowed. A name without a value is automatically assigned
an increment of the value of the previous name. It shall be an error to automatically increment the maximum
representable value of the enum.
// c is automatically assigned the increment-value of 8
enum {a=3, b=7, c} alphabet;
Both the enumeration names and their integer values shall be unique. It shall be an error to set two values to
the same name or to set the same value to two names, regardless of whether the values are set explicitly or
by automatic incrementing.
// Error: c and d are both assigned 8
enum {a=0, b=7, c, d=8} alphabet;
If the first name is not assigned a value, it is given the initial value of 0.
// a=0, b=7, c=8
enum {a, b=7, c} alphabet;

alaindargelas, 2025-06-18:
Fixed by #4041

closed 2025-06-18 by alaindargelas