chipsalliance/Surelog #4017: Error using port connection with dot notation and `inout`

tcal-x, 2024-11-21:
Contents of test file `InoutConnect.sv`:
```verilog
module InoutConnect(
                    .X1(internal), 
                    .X2(internal)
                    );

   parameter width = 1;

   inout [ width - 1 : 0 ] internal;
endmodule // InoutConnect
```
How to run:
```
% synlig
synlig> read_systemverilog InoutConnect.sv
```
Output:
```
1. Executing SystemVerilog frontend.
[INF:CM0023] Creating log file "/home/tim/rtl/slpp_all/surelog.log".
[INF:CP0300] Compilation...
[INF:CP0303] /home/tim/rtl/InoutConnect.sv:2:1: Compile module "work@InoutConnect".
[INF:CP0302] Compile class "work@mailbox".
[INF:CP0302] Compile class "work@process".
[INF:CP0302] Compile class "work@semaphore".
[NTE:CP0309] /home/tim/rtl/InoutConnect.sv:10:28: Implicit port type (wire) for "internal".
[INF:EL0526] Design Elaboration...
[NTE:EL0503] /home/tim/rtl/InoutConnect.sv:2:1: Top level module "work@InoutConnect".
[NTE:EL0508] Nb Top level modules: 1.
[NTE:EL0509] Max instance depth: 1.
[NTE:EL0510] Nb instances: 1.
[NTE:EL0511] Nb leaf instances: 1.
[INF:UH0706] Creating UHDM Model...
[INF:UH0707] Elaborating UHDM...
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 6
VPI ERROR: Bad usage of vpi_get
Segmentation fault (core dumped)
```

tcal-x, 2024-11-21:
Additional information in `./slpp_all/surelog.log`:
```
VERSION: 1.84
BUILT  : Nov 20 2024
DATE   : 2024-11-21.11:47:19
COMMAND: InoutConnect.sv -DYOSYS=1 -DSYNTHESIS=1
```

alaindargelas, 2025-06-19:
This is a non-ANSI module port declaration, LRM example:
Example 4: Two ports with different names connected to same internal net (non-ANSI style module header)
module same_port (.a(i), .b(i));
 // Name 'i' is declared inside the module as an inout port.
 // Names 'a' and 'b' are defined for port connections.

alaindargelas, 2025-06-25:
Fixed by #4042

closed 2025-06-25 by alaindargelas