chipsalliance/Surelog #4032: Incorrect ref_obj resolution for port connections when the actual net name is the same as the port name

mlr11, 2025-05-12:
In the example below:

```
module dut_with_array_net (
    input  logic in_array [3:0],
    output logic out_array [3:0]
);
    // Direct assignments, no generate
    assign out_array[0] = in_array[0];
    assign out_array[1] = in_array[1];
    assign out_array[2] = in_array[2];
    assign out_array[3] = in_array[3];
endmodule

module tb_array_net;
    logic in_array [3:0];
    logic out_array [3:0];

    dut_with_array_net dut (
        .in_array(in_array),
        .out_array(out_array)
    );
endmodule
```

The following UHDM is created (after elaboration with Surelog & UHDM)

```
    |vpiInstance:
    \_module_inst: work@tb_array_net (work@tb_array_net), file:/home/mario/Projects/cayenne/tests/arraynet2/arraynet2.sv, line:12:1, endln:30:10
    |vpiPort:
    \_port: (in_array), line:2:18, endln:2:26
      |vpiParent:
      \_module_inst: work@dut_with_array_net (work@tb_array_net.dut), file:/home/mario/Projects/cayenne/tests/arraynet2/arraynet2.sv, line:16:5, endln:19:7
      |vpiName:in_array
      |vpiDirection:1
      |vpiHighConn:
      \_ref_obj: (work@tb_array_net.in_array), line:17:19, endln:17:27
        |vpiParent:
        \_port: (in_array), line:2:18, endln:2:26
        |vpiName:in_array
        |vpiFullName:work@tb_array_net.in_array
        |vpiActual:
        \_array_net: (work@tb_array_net.dut.in_array), line:2:18, endln:2:26
      |vpiLowConn:
      \_ref_obj: (work@tb_array_net.dut.in_array), line:17:10, endln:17:18
        |vpiParent:
        \_port: (in_array), line:2:18, endln:2:26
        |vpiName:in_array
        |vpiFullName:work@tb_array_net.dut.in_array
        |vpiActual:
        \_array_net: (work@tb_array_net.dut.in_array), line:2:18, endln:2:26
      |vpiTypedef:
      \_ref_typespec: (work@tb_array_net.dut.in_array)
        |vpiParent:
        \_port: (in_array), line:2:18, endln:2:26
        |vpiFullName:work@tb_array_net.dut.in_array
        |vpiActual:
        \_logic_typespec: , line:2:12, endln:2:17
      |vpiInstance:
      \_module_inst: work@dut_with_array_net (work@tb_array_net.dut), file:/home/mario/Projects/cayenne/tests/arraynet2/arraynet2.sv, line:16:5, endln:19:7
```
if you look at the vpiHighConn you'll see a ref_obj to work@tb_array_net.in_array and the actual being work@tb_array_net.dut.in_array.
This resolves to the net inside the dut instead of the net in the testbench.

Updating the names for the same example to avoid the testbench net name to be the same as the port name.

```
module dut_with_array_net (
    input  logic in_array [3:0],
    output logic out_array [3:0]
);
    // Direct assignments, no generate
    assign out_array[0] = in_array[0];
    assign out_array[1] = in_array[1];
    assign out_array[2] = in_array[2];
    assign out_array[3] = in_array[3];
endmodule

module tb_array_net;
    logic tb_in_array [3:0];
    logic tb_out_array [3:0];

    dut_with_array_net dut (
        .in_array(tb_in_array),
        .out_array(tb_out_array)
    );
endmodule
```
produces a slightly different UHDM (this is the expected result I believe).

```
    |vpiInstance:
    \_module_inst: work@tb_array_net (work@tb_array_net), file:/home/mario/Projects/cayenne/tests/arraynet/arraynet.sv, line:15:1, endln:33:10
    |vpiPort:
    \_port: (in_array), line:4:18, endln:4:26
      |vpiParent:
      \_module_inst: work@dut_with_array_net (work@tb_array_net.dut), file:/home/mario/Projects/cayenne/tests/arraynet/arraynet.sv, line:19:5, endln:22:7
      |vpiName:in_array
      |vpiDirection:1
      |vpiHighConn:
      \_ref_obj: (work@tb_array_net.tb_in_array), line:20:19, endln:20:30
        |vpiParent:
        \_port: (in_array), line:4:18, endln:4:26
        |vpiName:tb_in_array
        |vpiFullName:work@tb_array_net.tb_in_array
        |vpiActual:
        \_array_var: (work@tb_array_net.tb_in_array), line:16:11, endln:16:22
      |vpiLowConn:
      \_ref_obj: (work@tb_array_net.dut.in_array), line:20:10, endln:20:18
        |vpiParent:
        \_port: (in_array), line:4:18, endln:4:26
        |vpiName:in_array
        |vpiFullName:work@tb_array_net.dut.in_array
        |vpiActual:
        \_array_net: (work@tb_array_net.dut.in_array), line:4:18, endln:4:26
      |vpiTypedef:
      \_ref_typespec: (work@tb_array_net.dut.in_array)
        |vpiParent:
        \_port: (in_array), line:4:18, endln:4:26
        |vpiFullName:work@tb_array_net.dut.in_array
        |vpiActual:
        \_logic_typespec: , line:4:12, endln:4:17
      |vpiInstance:
      \_module_inst: work@dut_with_array_net (work@tb_array_net.dut), file:/home/mario/Projects/cayenne/tests/arraynet/arraynet.sv, line:19:5, endln:22:7
```

In this scenario, the vpiHighConn for in_array correctly resolves to work@tb_array_net.tb_in_array

It seems like when doing the lookup, if a net of the same name exists in the module being instantiated, it is automatically used.  The resolution is done correctly when the net name doesn't exist in the module being instantiated.

I tried the in_array testbench net to the out_array port and vice versa which resulted in connections being shorted.  dut.in_array being connected to dut.out_array which isn't the same in the module.

alaindargelas, 2025-06-12:
The problem is with the -elabuhdm option (Uhdm final elab), similar problem reported here: https://github.com/chipsalliance/UHDM/issues/1074

alaindargelas, 2025-06-13:
Fixed by #4038